FreeCAD/FreeCAD #12960: Disappearing body or blue box when trying to model a herringbone gear.

forflo, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I get the described behavior on 0.21.2.

## Recipe

 - Open a new document
 - Change into Part Design
 - Create Body
 - Klick on newly created Body
 - Go to "Part Design" -> Involute Gear
 - Configure the Involute Gear Form as follows
   -  7 teeth
   - a module of 2.5mm
 - Klick ok accepting the other default Gear parameters
 - Klick on the newly created "InvoluteGear"-Item that has been added to the object tree of the body
 - Create an additive helix with the following parameters
   - Axis: Base on Z-axis
   - Mode: Height-Turns-Angle
   - Height: 10mm
   - Turns: 0.1

- Use the mirror tool to mirror the helical gear on the XY-Plane of the Body
- Add a new sketch to the body on the YZ-Plane:
  [image]
- Constrain it fully or leave it as in the picture (does not matter)
- Use the new sketch with the rectangle to create a revolution around the base Z-axis.
- The end result should look like
  [image]

## The problem

- Although the last step in `Recipe` produces the desired revolution, when you try to edit the sketch that the revolution is based on, you get a blue rectangle that covers the entire body (as shown below) or the body disappears and the sketch looks somewhat broken.
  [image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-14:
Very specific problem report. Can you test this in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and also upload the file as a .zip?

forflo, 2024-03-15:
I double checked, but I can't reliably reproduce the issue anymore. I am closing this for now.

closed 2024-03-15 by luzpaz

drcowsley, 2024-04-30:
I have a missing body anomaly with helix - in my case reducing the pitch below 25mm loses the body from the rendered image.   I need a pitch of 6mm.
I'm busy looking for a workaround but it wouldn't do any harm for somebody to be looking at the problem in the mean time.

DrC

[HelixAnomaly.zip](https://github.com/FreeCAD/FreeCAD/files/15171525/HelixAnomaly.zip)

[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]