FreeCAD/FreeCAD #12964: There is no HTML export option

pierreporte, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Inspired by #12843. 3D PDF isn’t really a cross-platform format. Web technologies are currently able to produce a file that has the same features that works in any modern browser: a single HTML file can contain 3D files and everything needed to display it and interact with it.

Solidworks eDrawing, a visualization and publisihing preparation companion program of Solidworks, have a feature to export 3D files (not just native Solidworks files) in various formats including a standalone HTML file. The generated file has a lot of features, including displaying exploded views, displaying the tree view of the model, defining section views, and making measurement.

A FreeCAD equivalent could start more modest with only the tree view and exploded views.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36331 (Git) AppImage
Build type: Release
Branch: main
Hash: eb5a1b333946ce4ac8a4ead835cba8bdcf602a87
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * TabBar (Disabled)
  * Curves 0.6.23
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.3
  * OpenDark 2023.12.1
  * OpenLight 2023.12.17
```

### Subproject(s) affected?

Core

### Anything else?

[image]

[image]

[image]

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-15:
@pierreporte there is already the option to export as WebGL / X3D file (.html or .htmlx).
Please test if this is sufficient and I'll close this issue

pierreporte, 2024-03-15:
@maxwxyz The current solutions are not sufficient and not very satisfactory overall.

First the two options generate quite a different result and it isn’t clear when selecting one or the other.

Second the HTML format (not xHTML) requires the user to manually select the bodies to export or there is an error message, while the other doesn’t.

Third, there is no tree view, except for the HTML that has a list of the bodies.

And last, there are no custom views.

I will edit the issue to transform it into an improvement of the existing features.

pierreporte, 2024-03-15:
@maxwxyz update done.

wwmayer, 2024-11-06:
See also: https://forum.freecad.org/viewtopic.php?t=91824&start=20

m1cha1, 2026-09-18:
I think it would be better if the WebGL export used [three-cad-viewer](https://www.jsdelivr.com/package/npm/three-cad-viewer). It is a WebGL-based CAD viewer built on Three.js and is released under the MIT 
[Live example](https://bernhard-42.github.io/three-cad-viewer/example.html)

Screenshot below:
[image]