FreeCAD/FreeCAD #12968: [FEATURE] Add Changelog feature for addon devs to display what is new in an addon update

luzpaz, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
Not necessarily. It would be nice as an end user to know what was updated in the latest addon that needs to be updated. 

**Describe the solution you'd like**
Perhaps a dialog box or some sort of output when installing an update of a specific addon to output what has changed in this version update.

**Describe alternatives you've considered**
Providing a link back to the repository somehow?

**Additional context**
Several addon devs provide changelogs in their READMEs. For example:  
* https://github.com/mwganson/Editor_Assistant#changelog
* https://github.com/Zolko-123/FreeCAD_Assembly4#release-notes
* https://github.com/shaise/FreeCAD_SheetMetal#release-notes
* https://github.com/shaise/FreeCAD_FastenersWB#release-notes-berberic2

hasecilu, 2024-09-26:
If this is really planned there should be some rules to follow in order to facilitate the implementation on FreeCAD.

I was reading about [Common Changelog](https://common-changelog.org) and I like it. I think we could make a `diff` between the `CHANGELOG.md` files comparing the current/installed and the new one.

Or maybe better,  can toggle between the README and the CHANGELOG in the same way as GitHub does.

luzpaz, 2024-09-26:
I think standardizing a bit more makes sense. We could enforce it through https://github.com/FreeCAD/FreeCAD-addons/issues/160