FreeCAD/FreeCAD #12970: Feature: Addon Manager (in the background) notifies of any addons that have updates

luzpaz, 2024-03-14:
Ability for the Addon Manager to signal to the user that there are updates pending for specific addons.

pinkavaj, 2024-09-13:
@luzpaz Why it should do it in the background (why it is not sufficient to check it during start)? I would like to hear more about what this will solve/improve.

luzpaz, 2024-09-14:
@pinkavaj i'm not remembering exactly the burning impulse I had that made this ticket happen. I can speculate that:  

* at the time many addons were being developed and fixed consistently. Therefore as users would be dealing with a forced workarounds for said addon, it would be a potential benefit that the app signal them that an update was available. Perhaps it could potentially make their workflow easier.
* Some users don't turn off their computers or close FreeCAD for long periods of time. 
* Instead of checking manually for available updates, the program would do it for them periodically.