FreeCAD/FreeCAD #12973: Addon Manager does not understand renaming an addon

luzpaz, 2024-03-15:
Recently I added Assembly4 workbench: 
I initially added it as 'A4': 6914f4874bd7227082d799a873fa87aa880ddd1c  
and then reverted: dacf4084d43373c0520d708c8fe1af0a1d77e91c  
(didn't remove the `A4/` directory though)  
then I re-added the workbench as Assembl4: 321fcde29d97c1e1d7088de84aa35de0a93a9bb3  
Later, kkremitzki removed the A4/ dir in: 4c2d603087a302800eec48576bc36b5969fbeb49

This caused some issues downstream for FC users: 
https://forum.freecadweb.org/viewtopic.php?f=4&t=39945  

We may need logic in the FC Addon Manager that removes/modifies local ~/.FreeCAD/Mod/ subdirectories if their upstream has changes as well. 

@yorikvanhavre just recently added a fix: https://github.com/FreeCAD/FreeCAD/commit/f66023a646  
~~Not sure if this takes care of the issue.~~  
Edit: the fix doesn't address the above issue

maxwxyz, 2024-08-01:
is this still an issue or fixed with the Addon Manager overhaul?

chennes, 2024-08-02:
Still an issue.