FreeCAD/FreeCAD #12978: Crashes on Mesa >0.24

RandomLegend, 2024-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Appimages for FreeCAD crash on Linux with AMD cards running Mesa >0.24 and require Mesa downgrade to 23.5 to run.

Error produced is:

```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Custom folder not found: /mnt/Wyrm/Nextcloud/Kreativ/Drucker, Laser, CNC/DruckerFailed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
/tmp/.mount_FreeCAmhbENP/AppRun: line 43: 77083 Aborted                 (core dumped) ${MAIN} "$@"
```

And this happens on all forks and versions. Whether it be the stable version, dev weekly build or the Ondsel fork.

### Full version info

```shell
Can't provide that info as the application doesn't start.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-15:
I think that's a specific problem of the AppImage because it contains driver libs. I have seen the same with an AppImage for v0.17 or so. After unpacking and removing these libraries the problem was solved.

RandomLegend, 2024-03-17:
Can you explain me how i could unpack the appimage and remove the libraries?

wwmayer, 2024-03-17:
https://docs.appimage.org/user-guide/run-appimages.html#id5

Alternatively, you can start the AppImage and then check its mount point which usually is a sub-directory in /tmp. Copy this directory somewhere else.

The graphic drivers may have the term "dri" in their filenames.

RandomLegend, 2024-03-17:
I extracted the appimage but i can't find anything in it's /usr/lib about /dri

The error message shows me /usr/lib/dri/radeonsi_dri.so and swrast_dri.so but there isn't even a /dri directory that i can find

wwmayer, 2024-03-17:
Which exact AppImage have you downloaded?

carsten-bruckhaus, 2024-03-18:
The problem was also reported in realthunder's branch and a solution/workaround posted, [that I can confirm works for me](https://github.com/realthunder/FreeCAD/issues/960#issuecomment-1974233355).

Apparently there are `libdrm*` files included in the AppImage, that interfere with system libraries. Removing those and repacking seems to solve the problem.

oerkel47, 2024-03-20:
Fyi I think the fix is already and in the daily builds for some time.

nazar-pc, 2024-06-29:
Just hit this with stable on fresh Ubuntu 24.04 with Mesa from git (through oibaf ppa), while dev build works fine on the same system:
```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

failed to load driver: radeonsi
failed to load driver: radeonsi
failed to load driver: radeonsi
failed to load driver: radeonsi
QApplication: invalid style override 'Adwaita-Dark' passed, ignoring it.
	Available styles: Windows, Fusion
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_FreeCA1js0Mf/AppRun: line 43: 1593692 Aborted                 (core dumped) ${MAIN} "$@"
```