FreeCAD/FreeCAD #12979: Status bar is not customizable.

PaddleStroke, 2024-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Linkstage has this cool feature where users can customize the status bar : 
[image]
It is possible to : 
- Show/hide each widget of the statusbar.
- Drop toolbars in the statusbar.

From discussion in https://github.com/FreeCAD/FreeCAD/issues/12742 : 

> The toolbar drag and drop on to status bar is implemented in this [commit](https://github.com/realthunder/FreeCAD/commit/d15bc174739). This relies on QStatusBar.addWidget(). There is no similar API for title bar. In fact, title bars are supplied by native UI platform. You'll need write a completely new widget to replace the title bar.

@realthunder Could you please make a PR to port this feature to core? I'm having a look at the commit, but it seems core and linkstage had already diverged significantly.
As a bonus if you can also put such 'docking areas' in the menubar left and right corner widgets? Currently the corner widgets are used by an option to put the workbench selector there. But if we had docking areas, then we can remove this specific option and let users directly drop the workbench selector in the menubar 'docking area'.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-16:
I'll give it a try soon.

PaddleStroke, 2024-04-15:
@realthunder any update on this ? :)

realthunder, 2024-04-15:
I've done it in my branch. Haven't got time to split it out for master. I'll submit PR soon.