FreeCAD/FreeCAD #12980: Core doesn't have datums (coordinates system, planes...)

PaddleStroke, 2024-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Core does not have datum commands : 
- Coordinate system
- Plane
- Line
- Point

It would be usefull that these were available in the Structure core toolbar, rather than implemented by each workbench that needs them (part design, a4, MBD, ...)
I have seen that linkstage does have this feature : 
[image]

Any reaction to this @wwmayer ?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-15:
Core has App::Line, App::Plane, App::Origin (as coordinate system)

maxwxyz, 2024-03-15:
@PaddleStroke is this any different to https://github.com/FreeCAD/FreeCAD/issues/11607 or https://github.com/FreeCAD/FreeCAD/issues/12584

PaddleStroke, 2024-03-15:
Not related to 12584, but yes it's the same as 11607, just enunciated in a more global way.

maxwxyz, 2024-03-15:
Can I add your post to a comment there and close this one, or @pierreporte could you rephrase your issue to reflect the global use case?

pierreporte, 2024-03-15:
@maxwxyz I will edit my issue. It would indeed be great if tools were more generic and available more broadly. There is a similar problem with sketches though not as severe (see #12106).

maxwxyz, 2024-03-15:
Duplicate of #11607

closed 2024-03-15 by maxwxyz