FreeCAD/FreeCAD #12983: Assembly: Joints do not adjust when bodies underlying assembly parts change

johnsonm, 2024-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

 I have a set of bodies driven by a configuration table in one document, and an assembly comprised of only one grounded part and multiple fixed parts in another document. When I change the configuration of the bodies, the shapes of the parts in the assembly change correctly, but their relationships do not; the solver doesn't appear to re-run, even if I try to recompute the assembly. My goal was to use the assembly to validate all the configuration made sense, but so far it's not working for that.

The different configurations are different only in size; they don't change the parts topologically.

If I delete all the Fixed joints and recreate them, they are assembled correctly, until the next time I choose a different configuration.

### Full version info

```shell
OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36331 (Git) AppImage
Build type: Release
Branch: main
Hash: eb5a1b333946ce4ac8a4ead835cba8bdcf602a87
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
```

### Subproject(s) affected?

Assembly

### Anything else?

Uploading a .zip of the .FCStd files failed without a clear error from github, so here are links to two files that demonstrate this behavior:

https://gitlab.com/mcdanlj/FloatingRotaryBroach/-/blob/e37cada7d0ad7dfe49baeca4ccfd788625e45ebf/FloatingRotaryBroach.FCStd
https://gitlab.com/mcdanlj/FloatingRotaryBroach/-/blob/3be1c5fa85f7162170ac189b5819fa670687f6dc/FloatingRotaryBroachAssembly.FCStd

In the first file, select the **Broach**, change the Size to another selection, and refresh. The set of parts (overlapped meaninglessly in that file) should change. In the assembly, the parts also change, but their relationships will not be adjusted following the different sizes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-21:
I have just tested. It appears that recomputing the assembly does indeed nothing.
But clicking on the "solve assembly" button does work.

So what is missing is that the recompute of the assembly object should trigger a solve of the assembly.

PaddleStroke, 2024-03-21:
Calling solve on recompute is fairly straight forward by reimplementing execute like so : 
App::DocumentObjectExecReturn* AssemblyObject::execute()
{
    App::DocumentObjectExecReturn* ret = App::Part::execute();
    solve();
    return ret;
}

My issue is that doing so modify the placement of the assembly objects. And this triggers warnings : 

> 15:35:54  539.724 <App> Document.cpp(2887): __cubes#Box001 still touched after recompute
> 15:35:54  539.724 <App> Document.cpp(2887): __cubes#Box002 still touched after recompute
> 15:35:54  539.724 <App> Document.cpp(2887): __cubes#Fixed still touched after recompute

@wwmayer Any idea how we could deal with this situation?

johnsonm, 2024-03-21:
I admit I didn't see the "solve assembly" button, so perhaps PEBCAK. I found the same option in Assembly3, so I don't know why I didn't see it in the integrated assembly. :flushed: 

But I guess I demonstrated that a naive user might expect recalculating to also solve the assembly. This seems like a potential UI improvement if there is a clear way, but lower priority knowing that I just didn't find an important button.

Thank you!

PaddleStroke, 2024-03-22:
I think I would expect the same as you.
Having said that calling solve on each recompute seems to come with its own challenge. It may have some unwanted consequences (see mentioned above for a start).

wwmayer, 2024-03-22:
> My issue is that doing so modify the placement of the assembly objects. And this triggers warnings :

This is caused by the solve() method. If I press the Solve button without having changed anything then a lot of objects get touched.

> Any idea how we could deal with this situation?

You have to understand what is happening. When you recompute a document then the objects are sorted topologically so that objects that do not depend on other objects are recomputed, then objects that depend on already recomputed object  and so on. The very last object will be the Assembly container.

But calling the solve() method for the Assembly again touches properties of the already recomputed objects, hence the warnings.

You have two alternatives to handle the problem:
1. Change the solve() method to not touch other objects if actually nothing has changed
2. If this is too difficult to achieve but you know that all objects are in a correct state you can simply purge the "touched"  flag of the affected objects

PaddleStroke, 2024-03-25:
Thanks for the feedback Werner. Solve only changes the placement of the objects, which does not seem to require a recompute. So I think the touched flag can be purged.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre