FreeCAD/FreeCAD #12984: Doing Ctrl-Z in TechDraw reliably causes a segfault

DavidGriffith, 2024-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While trying to figure out how to make multiple copies of a view in the TechDraw workbench, I would do something that didn't result in what I want, then when I hit Ctrl-Z to back out of it, FreeCAD segfaulted.  I first mentioned it in https://forum.freecad.org/viewtopic.php?p=747372#p747372.

To cause a segfault, load the attached file and do the following, which I copied from that post to forum.freecad.org:
````
- Select the view in the tree
- Ctrl-C
- click "Use Original Selections"
- paste with Ctrl-V

-> this places the new view on top level of the tree, not on the page

- select the page in the tree
- click on the value of the Views property
- click the ellipsis button
- add the copied view (or multiples of it)
- confirm 
````

Now do Ctrl-Z and see the crash.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (MATE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36181 (Git)
Build type: Release
Branch: main
Hash: f88f48f64f223085f97f0f2945298f9023d09693
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.2
  * sheetmetal 0.4.0
  * freecad.gears 1.0.0
  * Assembly4 0.50.7
  * A2plus 0.4.62
  * fasteners 0.5.1
  * Manipulator 1.5.7
```
```

### Subproject(s) affected?

None

### Anything else?

Load this for provoking a segfault:
[drilling-template.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14621564/drilling-template.FCStd.zip)

Crash messages:
````
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7fccefa5b050]
#1  0x7fcc307baf59 in TechDrawGui::ViewProviderViewPart::fixSceneDependencies() from /home/dave/proj/freecad/install/lib/TechDrawGui.so+0x19
#2  0x7fcc307a0446 in TechDrawGui::ViewProviderPage::fixSceneDependencies() from /home/dave/proj/freecad/install/lib/TechDrawGui.so+0x86
#3  0x7fcc30708682 in TechDrawGui::MDIViewPage::onMsg(char const*, char const**) from /home/dave/proj/freecad/install/lib/TechDrawGui.so+0x192
#4  0x7fccf2b47499 in Gui::Application::sendMsgToActiveView(char const*, char const**) from /opt/freecad/lib/libFreeCADGui.so+0x39
#5  0x7fccf2c662e1 in Gui::Command::_invoke(int, bool) from /opt/freecad/lib/libFreeCADGui.so+0x281
#6  0x7fccf2c66753 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /opt/freecad/lib/libFreeCADGui.so+0x133
#7  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f4f) [0x7fccefee8f4f]
#8  0x7fccf0b5c782 in QAction::triggered(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#9  0x7fccf0b5f3ab in QAction::activate(QAction::ActionEvent) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xab
#10  0x7fccf2cd437f in Gui::ShortcutManager::onTimer() from /opt/freecad/lib/libFreeCADGui.so+0xef
#11  0x7fccf2cd4eb4 in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /opt/freecad/lib/libFreeCADGui.so+0x4f4
#12  0x7fccf2cd5ef6 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /opt/freecad/lib/libFreeCADGui.so+0x66
#13  0x7fccefeb135e in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x7e
#14  0x7fccf0b62fe8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xb8
#15  0x7fccf2c0bf68 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/freecad/lib/libFreeCADGui.so+0x68
#16  0x7fccefeb16f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#17  0x7fccf036c76b in QShortcutMap::dispatchEvent(QKeyEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x44b
#18  0x7fccf036d0bb in QShortcutMap::tryShortcut(QKeyEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x5b
#19  0x7fccf0315376 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x176
#20  0x7fccf03390cd in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x8d
#21  0x7fccf0311cac in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#22  /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6deca) [0x7fccea17aeca]
#23  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7fccee11e7a9]
#24  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7fccee11ea38]
#25  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7fccee11eacc]
#26  0x7fcceff09836 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#27  0x7fccefeb017b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#28  0x7fccefeb82d6 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#29  0x7fccf2b54496 in Gui::Application::runApplication() from /opt/freecad/lib/libFreeCADGui.so+0x756
#30  FreeCAD(+0x8f0e) [0x5619ee2caf0e]
#31  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7fccefa4624a]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fccefa46305]
#33  FreeCAD(+0x9211) [0x5619ee2cb211]
````

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-16:
The crash happens inside ViewProviderViewPart::fixSceneDependencies() where the function getViewProviderPage() returns a null pointer. But this isn't checked and de-referencing causes the segmentation fault.

A fix is fairly easy:
```
void ViewProviderViewPart::fixSceneDependencies()
{
    auto page = getViewProviderPage();
    if (!page) {
        return;
    }

    auto scene = page->getQGSPage();
...
```

wwmayer, 2024-03-16:
https://github.com/FreeCAD/FreeCAD/pull/12986

commit f2f312ce (2024-03-16): TD: fixes #12984: Doing Ctrl-Z in TechDraw reliably causes a segfault
closed 2024-03-16 by wwmayer