FreeCAD/FreeCAD #12988: Curves WB doesnt run on FreeCad 0.22 Dev snap verrsion (probaly due to old numpy libray in snap package)

projectcad, 2024-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Curves WB doesnt run on FreeCad 0.22 Dev snap verrsion .....probaly due to old numpy libray in snap package

thanks in advance....

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36442 (Git) Snap 957
Build type: Release
Branch: main
Hash: bcf54d3153fe4cad841ff7c0b692506eaa802103
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-16:
is this related to the main FreeCAD or shouldn't this be fixed in the Curves WB repo?

luzpaz, 2024-03-16:
...or https://github.com/FreeCAD/FreeCAD-snap ?

projectcad, 2024-03-21:
i dont what actually was corrrected but it is ok after last update today

closed 2024-03-21 by projectcad

projectcad, 2024-03-21:
maybe this one corrected  ....https://github.com/FreeCAD/FreeCAD/commit/655d5e4fbd71053ebf3e37345d5c33e9cbee9f52