FreeCAD/FreeCAD #12990: Path grbl post processor produces unsupported canned drill cycles

nmw01223, 2024-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Grbl 1.1 - which seems the most common version - does not support canned drilling cycles. However, by default, that is what the post-processor will output.

One can add --translate_drill which alters the output back to G0/G1, but it seems undesirable that by default, it produces code that will not run. Especially as it is not at all obvious without fortuitously finding the information that this is what you need to do.

As there is already an option to turn canned drilling on, being --notranslate_drill, all that would have to be done is to alter the initial default in the code.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-31:
@nmw01223 can you retest on latest 1.1dev[^1] ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

PaxTux, 2025-07-28:
The behaviour is still the same in latest 1.1.0dev (commit No.: 42635).
By default grbl postprocessor in FreeCAD creates drilling codes that are not supported by grbl.

This topic was already discussed [here](https://github.com/FreeCAD/FreeCAD/pull/2255) and decision was made that the default option is `TRANSLATE_DRILL_CYCLES = False`.

Decision was made based on info that "any g-code feeder dealing with grbl" will translate unsupported codes like G81/G83...
However there are many who don't do that. In [this topic](https://github.com/gnea/grbl/issues/290) one user said that grbl-panel, Candle and UGS don't translate it. Topic is old though, I'm not sure what the currect status is, but I know Candle still cannot do this.

I think this decision should be reconsidered. Otherwise many more people will be scratching their heads (like I did once) why their machine doens't perform drilling operations.

J-Dunn, 2025-08-13:
> Decision was made based on info that "any g-code feeder dealing with grbl" will translate unsupported codes like G81/G83...

That seems perverse. Like , it's not my fault, let him do it. 
A post processor specifically dedicated to processing FreeCAD output for GRBL interpretors , clearly should do just that. 
On the contrary I would not expect a feeder to do more than feed.  I use UGS. It feeds, it does not expand drilling codes. 

The post proc should do it's job to the best of it's ability. If you don't like the result and think your feeder can do it better , by all means turn off grbl_proc drill expansion.

LarryWoestman, 2025-10-22:
You don't want to do this for all of the refactored posts, just the refactored grbl post (for now).  Otherwise you will translate drilling for posts and controllers that don't need it.

In the refactored grbl post file:

Add `values["TRANSLATE_DRILL_CYCLES"] = True` to the end of the "init_values" function (or, if you want to be picky, before the "USE_TLO" value and comment in that function so everything is in alphabetical order).  This changes the default value for translating drilling to "do it" rather than "don't do it".

Add `argument_defaults["translate_drill"] = True` to the end of the "init_argument_defaults" function.  This changes the tool tip to show "Translate drill cycles..." is the default instead of "Don't translate drill cycles...".

Warn people that you have just changed the way that the refactored grbl post works by default :-)

closed 2025-11-28 by sliptonic