FreeCAD/FreeCAD #12997: Qt6 build failing ‘class QStyleOptionButton’ has no member named ‘init’

mark-herbert42, 2024-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

src/Tools/plugins/widget/customwidgets.cpp:1836:9: error: ‘class QStyleOptionButton’ has no member named ‘init’

### Full version info

```shell
Seems like API has changed. This file is the only place wheere QStyleOptionButton is used in FreeCAD. 

Used git snapshot for compile
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-16:
@mark-herbert42 could you provide more info? On what system and how did you try building?

mark-herbert42, 2024-03-16:
Gentoo, ebuild freecad-9999, loading the git snaphot directly before build. Before todays patches fixing QRegExp it was failing before reaching this point.

wwmayer, 2024-03-16:
https://github.com/FreeCAD/FreeCAD/pull/13004

closed 2024-03-16 by wwmayer