FreeCAD/FreeCAD #13002: Validate Sketch should be available in Sketcher workbench

prokoudine, 2024-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the only way to check a sketch for errors is to exit Sketcher, select the sketch while in PD, then choose Sketch > Validate Sketch. It should be possible to validate a sketch without exiting Sketcher.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.36135 (Git) AppImage
Build type: Release
Branch: flavor
Hash: 5d3a33067a8c5073b468df07383ffa5028d1c065
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * Ondsel-Lens 2024.1.22.01
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-16:
The validation task panel could b a separate tab, as described in #12814.

maxwxyz, 2024-03-30:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/6261

maxwxyz, 2024-03-30:
https://forum.freecad.org/viewtopic.php?p=534845#p534889
Adding this from the imported issue.

benj5378, 2024-10-06:
[image]

Looks to me like the feature is available in Sketcher Workbench itself. Close this issue? @luzpaz

pierreporte, 2024-10-06:
@benj5378 It is not available while editing the sketch—that’s the problem.