FreeCAD/FreeCAD #13003: Sketcher Tool: Sketcher-Translate useless after Change

Dreisi, 2024-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the 0.22 version of FreeCad the function "Sketcher-Translate" (Matrix-Transformierung in German) has ben basicaly reworked.

Funktion is good, number of lines and colums where asked, x and y spacing and angle has to be set by mouse in the sketch. 

But after operating the multiplied objects are not linked by dependencies. In that way, the creation of an 8x3 Array of holes generates 52 new degress of freedom. I think this is useless. 

[image]
[image]
[image]

### Full version info

```shell
(OS: Debian GNU/Linux 12 (bookworm) (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36438 (Git)
Build type: Unknown
Branch: main
Hash: 0def330a9e2119a14bb349cc406c2aa365421ab4
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.56a
  * Assembly4 0.12.3
  * Assembly3 0.11.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-16:
So the idea is to create constraints for each of the copy, defining vertical and horizontal distance? The other constraints should be copied if selected.

Dreisi, 2024-04-02:
Hello, 

exact that is the Idea. In older versions the behavior was in that way. 
Disadvantage was that the sketcher went very slow wen many objects with many 
constrains hat to been moved or edited in other way.

In the screenshot there is an example. When I make 12 Holes in a 3x4 Array 
from one hole, I get 24 degrees of freedom. For further work I must set draw 
the contrains, set them all equal in x and/or y direction, bevor I can change 
distances or lock them to their final place. 

The Button "Randbedingungen klonen" dosn't work. It is locked in the disabled 
state. 

MfG Christian Kauer

Am Samstag, 16. März 2024, 21:06:35 CEST schrieb Max Wilfinger:
> So the idea is to create constraints for each of the copy, defining vertical
> and horizontal distance? The other constraints should be copied if
> selected.

-- 
-------------------------------------------
Dr. Christian Kauer
Perlsteinring 17
46284 Dorsten

Mobil: 015118400104

PaddleStroke, 2024-04-02:
If I understand correctly, you are asking for the same behavior as the rectangular array tool, that the tool creates construction lines between each copy ?

Dreisi, 2024-04-04:
Yes exactly, in  0.22.0dev.36438 (Git) the rectangular array tool ist removed
(oder hidden somewehere), so I thougt that matrix tool should replace it.

Am Dienstag, 2. April 2024, 09:29:01 CEST schrieb PaddleStroke:
> If I understand correctly, you are asking for the same behavior as the
> rectangular array tool, that the tool creates construction lines between
> each copy ?

--
-------------------------------------------
Dr. Christian Kauer
Perlsteinring 17
46284 Dorsten

Mobil: 015118400104