FreeCAD/FreeCAD #13008: Sketcher: Tool parameters focus issue (see transform tool)

PaddleStroke, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is a severe focus bug in sketcher's tool widgets, where clicking on the spinboxes have erratic results.
It is especially visible in transform tool when you try to interact with the spinboxes.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-17:
related to https://github.com/FreeCAD/FreeCAD/issues/12495 ?

PaddleStroke, 2024-03-17:
I think #12495 is probably fixed by my opened PR. Though I am not sure I
can't test right now.

FEA-eng, 2024-06-29:
Is this still relevant? What are the exact steps to reproduce?

closed 2024-07-01 by PaddleStroke