FreeCAD/FreeCAD #13009: External geometry is not removed

kaiwas, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In version 0.22 (dated March 11, 2024), external geometry is not deleted when making multiple selections.
But it can be removed by selecting each element with a single click.
However, not all selected items are displayed in the selection filter.
Video demonstration.

https://github.com/FreeCAD/FreeCAD/assets/32273833/d4a5eb94-dfc6-4b81-85a0-911e59d88ed0

### Full version info

```shell
0,22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-17:
can reproduce on
<details><summary>0.22dev Windows 11</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.14
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.3

``` 

</p>
</details> 

Not possible to delete either in the 3D view when hitting del-key.

maxwxyz, 2024-03-29:
@PaddleStroke FYI

PaddleStroke, 2024-03-29:
I would guess this comes from the TNP work as well. Probably an index issue, as you can see in the video when he selects one of the edge it does not get selected in the widget. Though this is only guess work. Any idea when this got introduced?

maxwxyz, 2024-03-29:
@bgbsww maybe any idea?

CalligaroV, 2024-04-04:
Hi Everyone!

I gave a look at this issue and yes, it's related to some modifications introduced for TNP.

For now I found a fix (probably temporary) that should solve both this issue and #13248.
The full fix is more complicated and needs to import more code from RT fork, I already started working on that.

Regardless for that, I opened #13297 that contains the temporary fix.
Can you try it? On my local build it works.

TIA!

kaiwas, 2024-06-16:
Open again (((
https://github.com/FreeCAD/FreeCAD/pull/13297#issuecomment-2171414622

FEA-eng, 2024-06-25:
It seems to be working now with #14437 merged.

PR #13297 (2024-04-05): App/Toponaming: GeoFeature - fixes #13009 - fixes #13248
 * Restored previous logic of GeoFeature::getElementName()
 * Added precompiler directive to use the old logic if FC_USE_TNP_FIX isn't defined, otherwise use the new logic
 * Reworked SketchObjectTest::getElementName() as a consequence of the previous points

fixes #13009
fixes #13248
commit 6779c912 (2024-04-05): App/Toponaming: GeoFeature - fixes #13009 - fixes #13248
 * Restored previous logic of GeoFeature::getElementName()
 * Added precompiler directive to use the old logic if FC_USE_TNP_FIX isn't defined, otherwise use the new logic
 * Reworked SketchObjectTest::getElementName() as a consequence of the previous points