FreeCAD/FreeCAD #13010: undeclared identifiers in DlgSettingsTheme::populateStylesheets

scorp08, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello
I build except freeCADGUI  results with undeclared identifiers in DlgSettingsTheme.cpp - QT5 with msvc 2017 - Release. Log is attached.
Kind regards to the community.

### Full version info

```shell
[cmake] -- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.19045.
[cmake] -- Install configuration: "Release"
[cmake] -- Compiler: MSVC, version: 19.16.27038.0
[cmake] -- prefix: D:/Freecadclone/out/install/x64
[cmake] -- bindir: bin
[cmake] -- datadir: data
[cmake] -- docdir: doc
[cmake] -- includedir: include
[cmake] -- libdir: lib
[cmake] -- cmake: 3.29.0-rc3
[cmake] -- Using Qt5 directory from LibPack in D:/Freecadclone/LibPack/lib/cmake/Qt5
[cmake] -- Compiling with Qt 5
[cmake] -- PyCXX found:
[cmake] --   Headers:  D:/Freecadclone/src
[cmake] --   Sources:  D:/Freecadclone/src/CXX
[cmake] --   Version:  7.1.9
[cmake] -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
[cmake] =====================================================
[cmake] Doxygen not found, will not build documentation.     
[cmake] =====================================================
[cmake] 
[cmake] -- Checking for connection to GitHub...
[cmake] -- GitHub connection established for FetchContent
[cmake] -- Module support is disabled.
[cmake] -- Version: 9.1.0
[cmake] -- Build type: 
[cmake] -- CXX_STANDARD: 17
[cmake] -- Required features: cxx_variadic_templates
[cmake] -- fmt was downloaded using FetchContent into D:/Freecadclone/out/build/x64/_deps/fmt-src
[cmake] -- PyCXX found:
[cmake] --   Headers:  D:/Freecadclone/src
[cmake] --   Sources:  D:/Freecadclone/src/CXX
[cmake] --   Version:  7.1.9
[cmake] -- -- Found OCE/OpenCASCADE version: 7.6.3
[cmake] -- -- OCE/OpenCASCADE include directory: D:/Freecadclone/LibPack/inc
[cmake] -- -- OCE/OpenCASCADE shared libraries directory: D:/Freecadclone/LibPack/lib
[cmake] -- HUHUH: D:/Freecadclone/LibPack/lib/cmake/vtk-8.2/VTKTargets.cmake
[cmake] -- VTK components: vtkCommonCore;vtkCommonDataModel;vtkFiltersVerdict;vtkIOXML;vtkFiltersCore;vtkFiltersGeneral;vtkIOLegacy;vtkFiltersExtraction;vtkFiltersSources;vtkFiltersGeometry;vtkhdf5;vtkRenderingCore;vtkInteractionStyle;vtkRenderingFreeType;vtkRenderingOpenGL2
[cmake] -- HUHUH: D:/Freecadclone/LibPack/lib/cmake/vtk-8.2/VTKTargets.cmake
[cmake] -- Check for medfile (libmed and libmedc) ...
[cmake] -- Found Boost: D:/Freecadclone/LibPack/include (found suitable version "1.67.0", minimum required is "1.65") found components: filesystem program_options regex system thread date_time chrono atomic
[cmake] -- Found Netgen: D:/Freecadclone/LibPack/cmake
[cmake] -- Found NETGEN version 6.2, calculated: 395833
[cmake] -- Set up to compile with Qt 5.15.2
[cmake] -- Shiboken2Config: Using default python: .cp38-win_amd64
[cmake] -- SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: 'D:/Freecadclone/LibPack/bin/include'
[cmake] -- SHIBOKEN_PYTHON_LIBRARIES computed to value: ''
[cmake] -- libshiboken built for Release
[cmake] -- PySide 5.15.0 Python module found at D:\Freecadclone\LibPack\bin\lib\site-packages\PySide2.
[cmake] 
[cmake] -- Found Matplotlib: D:/Freecadclone/LibPack/bin/lib/site-packages/matplotlib (found version "3.3.3")
[cmake] -- Platform is 64-bit, set -D_OCC64
[cmake] -- [E57] Revison ID: E57Format-2.2.1-AMD64_64-windows
[cmake] -- [E57] Building static library
[cmake] -- [OndselSolver] Building static library
[cmake] -- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) 
[cmake] -- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) 
[... 88 lines cut ...]
[cmake] -- QtNetwork:                   5.15.2
[cmake] -- QtOpenGL:                    5.15.2
[cmake] -- QtPrintSupport:              5.15.2
[cmake] -- QtSvg:                       5.15.2
[cmake] -- QtTest:                      5.15.2
[cmake] -- QtUiTools:                   5.15.2
[cmake] -- QtWebEngineWidgets:          5.15.2
[cmake] -- QtWidgets:                   5.15.2
[cmake] -- QtXml:                       5.15.2
[cmake] -- Rift:                        not enabled (BUILD_VR)
[cmake] -- Shiboken:                    5.15.0 [D:/Freecadclone/LibPack/include/shiboken2;D:/Freecadclone/LibPack/bin/include]
[cmake] -- SMESH:                       7.7.1.0 build internal
[cmake] -- SWIG:                        4.0.2
[cmake] -- VTK:                         8.2.0
[cmake] -- XercesC:                     3.2.2 [optimized;D:/Freecadclone/LibPack/lib/xerces-c_3.lib;debug;D:/Freecadclone/LibPack/lib/xerces-c_3D.lib] [D:/Freecadclone/LibPack/include]
[cmake] -- ZLIB:                        1.2.11
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

scorp08, 2024-03-19:
```
void DlgSettingsTheme::populateStylesheets(const char *key,
                                        const char *path,
                                        Gui::PrefComboBox *combo, // added gui namespace solved 
                                        const char *def,
                                        QStringList filter)
```

luzpaz, 2024-08-24:
@scorp08 can  you recreate on latest HEAD ? Does this also happen with Qt6 build ?

closed 2025-06-29 by github-actions[bot]