FreeCAD/FreeCAD #13013: Sketch transformation  tools  break the geometry

kaiwas, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

New sketch transformation tools (scale, motion, rotaid) - break the geometry. Those. previously assigned constraints at the junctions of lines are destroyed.

https://github.com/FreeCAD/FreeCAD/assets/32273833/cf7f1864-68f6-45bd-8d50-7806e56b7b58

Perhaps it would be more correct to preserve these constraints. Because restoring the continuity of geometry takes considerable time. In addition, it is a difficult task, because when assigning new connections, vertices can move when merging.
I consider the behavior shown in this video (blender) to be more logical.

https://github.com/FreeCAD/FreeCAD/assets/32273833/c560c5f1-9562-4904-8567-6514a7f53ca1

### Full version info

```shell
0,22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-17:
IMO it works as designed. If you need to move one point, just manually drag it or dimension it with constraints. The tools allow for much more then just moving a point: create copies, arrays,... 
If you want to rotate more than just one point, just select more geometry. All constraints inside the selected geometries will be kept and this does not *break* it.

kaiwas, 2024-03-17:
Quite a simple counter question. Have you tried to apply this logic to solve specific problems?
The configured constraints are a previously performed action. Any rupture or breakage of these restraints may be considered an error.
It’s more important to understand what is a priority. In my opinion, constraints are more important than action, because they are responsible for the final geometry.

maxwxyz, 2024-03-17:
I agree, this should be an additional option: to keep *external* constraints to the neighboring geometries which are not selected.

luzpaz, 2024-03-17:
rename the ticket then ?

PaddleStroke, 2024-03-18:
The tools were disgned to ignore constraints that connect to geos not selected. And for good reasons, I can easily give you examples where the behavior you want would result in conflicting constraints.
The scale tool is not intended to scale 1 or 2 edges of a closed shape. This should be done just by dragging the point or changing constraints. I can't see any real workflow where you would want to scale a single line of a closed wire.

kaiwas, 2024-03-18:
Not an easy task, considering that I don’t use freecad in my work because of its inconvenience (in its current form, but it looks like it may become even more inconvenient).

In general I understand your approach, but the changes you made affect two actions. In addition to movement, scale and rotation, one more action occurs “automatically” - breaking the geometry. If there was simply an option to disable this behavior, that would solve the problem. But you completely ignore that this could be useful.

From simple examples I can give this one.
Let's say we have a rectangle and we need to double it in one of the directions (along the Y axis).
If I select the entire rectangle and apply scale, it will change the whole thing (in all directions). And if I scaled one edge, I would get the desired result (scaling along only one of the axes).

[image]

Scale Y-axis in Corel Draw (easy and relaxed)
[image]

Scaling further breaks down the limitations of verticality and horizontality. And it completely breaks the geometry of the sketch. It takes some time to restore the geometry.

I wash hands. I will not prove anything further and will simply continue to monitor this topic. I am sure that sooner or later other people will have difficulties using new tools.
Just try solving DIFFERENT problems using different tools. (The same applies to various multitool innovations). Create a group of testers already (I’m still ready to participate).
Repairing the geometry after every movement is a dubious decision, but you can leave it if you have compelling arguments that repairing the geometry is a convenience.

kaiwas, 2024-03-18:
> I can easily give you examples where the behavior you want would result in conflicting constraints.

Can you give such an example? Very interesting

PaddleStroke, 2024-03-18:
[image]
Scale the selected line while preserving the coincidences.

kaiwas, 2024-03-18:
I expected something similar.
What if we temporarily convert the red size to blue? do a transformation and make it red again?
The entire rectangle will increase along the yoke axis.

Just two steps. Much easier than fixing broken geometry. In addition, I showed that scaling along the game axis can be a task of such scaling.

In your approach, in order to maintain one constraint, you destroy the other two. Funny.
Eventually, one could issue a warning that "the operation cannot be performed because constraint 'A' restricts this action." Come up with something like this, and not break the geometry.

Analogue in version 0.21 (you know there is no scaling)
[image]

kaiwas, 2024-03-19:
> Scale the selected line while preserving the coincidences.

I created a forum thread to discuss the behavior of these tools.
https://forum.freecad.org/viewtopic.php?t=86221

PaddleStroke, 2024-03-19:
First it would be very difficult to implement. How do you generalize this case of a single dimension?
What geometries do you scale then when selecting a single edge? All connected geometries? What about construction geometries? What about dimensions that are connecting the closed shape to another geometry then? What if there's no dimensions in the closed wire, but other constraints to another unconnected geo?
[image]

On top of that there is the problem that the tool acts in an imprevisible way.

Lastly I can't say I'm seeing any workflow where this would be useful in cad.

I can't say I'm 100% sure (because the feature request as it stands is not very clearly defined) but I would say that this can't be done and that it is not a reasonable feature request.

kaiwas, 2024-03-19:
> Lastly I can't say I'm seeing any workflow where this would be useful in cad.

I understand. But it sounds almost the same as "these tools are not required."
I don’t know how it works in other CAD (I only worked in Archicad and AutoCAD and there are no parametrics, sketches or constraints there).
But if we all worked together to figure out how this could work, perhaps it would be a revolution (if this is not the case in other ring roads).

Moving, rotating and scaling operations are some of the most common. And just as often they require maintaining the original proportions (preserving previously completed work). Now there is a great chance to break the geometry and then restore everything for a very long and painful time. IMO