FreeCAD/FreeCAD #13016: Part Design / UI: hard to identify tip position

maxwxyz, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the Part Design WB it is hard to identify where the current tip is.
It is currently indicated by a hardcoded .xpm which is small and pixelated, especially on small screens:
[image]

An option would be to underline the text of the feature the current tip is on, or an arrow next to the feature in the tree.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.14
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-17:
There have been a discussion in the DWG space on Discord.

Active body underlined and an arrow for the tip:

[image]

Radio button column:

[image]

Draggable line:

[image]

NomAnor, 2024-03-17:
I also think it would be better to have fewer overlay icons. The arrow looks fine, but i would align it on the right, so it is not influenced by the label length, then it's easier to find.

maxwxyz, 2024-03-17:
we could also use on all following icons/features the suppressed UI.

pierreporte, 2024-04-22:
@maxwxyz for the UI/UX project.

pierreporte, 2024-07-05:
Mockup by @kadet1090:

[image]