FreeCAD/FreeCAD #13017: Better Surface with Edges mode of the FEM results pipeline

FEA-eng, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The _Surface with Edges_ mode of the FEM results pipeline shows this:

[image]

I really don't like the poorly visible gray element edges. In ParaView, the edge color can be changed:

[image]

But you may also notice that ParaView shows the actual solid mesh which was generated and used for FEM analysis:

[image]

while FreeCAD displays a mesh with more triangles. This can be really misleading. It might be related to this: https://forum.freecad.org/viewtopic.php?t=78598 and thus hard to fix but maybe it's a less complex issue.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-17:
@FEA-eng I think they are currently not grey but the color of the label (changing color in your example). I think there is a way of displaying only edges and then you could not see the color if they are all black.

FEA-eng, 2024-03-17:
> @FEA-eng I think they are currently not grey but the color of the label (changing color in your example). I think there is a way of displaying only edges and then you could not see the color if they are all black.

Right, I mentioned grey because of this _None_ field mode:

[image]

But in contour plots, the edges somehow blend with the contours. They are way too thin and poorly visible.The edge-only (_Wireframe_) mode looks like this:

[image]

It shouldn't be black indeed. But in ParaView, the edge color setting applies only to the _Surface with Edges_ mode, the _Wireframe_ one looks like in FreeCAD and doesn't even have this setting displayed.

FEA-eng, 2024-03-18:
> We can add options to set wireframe line width, node point size, and a boolean 'PlainColorEdgeOnSurface' with the associated color property.

That would be ideal. But it would have to be added for all filters that can belong to the pipeline as well since they have the same rendering modes.

Do you know if it's possible to display the surface mesh properly (without additional triangles)?

marioalexis84, 2024-03-18:
>  But it would have to be added for all filters that can belong to the pipeline as well since they have the same rendering modes.

Do you mean that all the objects in the pipeline change the mode automatically according to the mode of the currently displayed object? Or that the proposed properties are available for all types of filters?

> Do you know if it's possible to display the surface mesh properly (without additional triangles)?

I'll try to solve it.

FEA-eng, 2024-03-18:
> Do you mean that all the objects in the pipeline change the mode automatically according to the mode of the currently displayed object? Or that the proposed properties are available for all types of filters?

The latter. Not for all filters but at least for most of them. Perhaps they should use the same settings as their parent pipeline by default. And maybe even the pipeline or the previous filter should be hidden automatically when a new filter is added. Because currently, you have to set the mode and field for each new filter and hide the previous ones to make it visible. But that's another topic.

> I'll try to solve it.

Thanks, that would be nice. It's nothing urgent but could improve postprocessing.

marioalexis84, 2024-03-18:
Properties added to the FemPostObject base class are inherited by all filters. Then we can discuss the default values.

marioalexis84, 2024-03-18:
That's another topic because Field values are specific to the solver and type of analysis. The core Fem objects should not know anything about the solver used.
I have a solution in mind for that, but I still have to implement the code.

commit 816226d3 (2024-04-01): Fem: Improve post-processing object style - fixes #13017
closed 2024-04-01 by chennes