FreeCAD/FreeCAD #13019: Material thermal expansion coefficient can be defined for FEM but not the reference temperature

FEA-eng, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Thermal material properties include the expansion coefficient but it should be also possible to specify the reference temperature (the temperature at which there are no thermal strains). CalculiX handles it this way:

```
*EXPANSION, ZERO=reference_temperature
expansion_coefficient
```
If not specified, it's assumed to be 0 which is often not the desired behavior.

It could be added as a separate material property to make it simple.

@davesrocketshop Can you add it at some point?

P.S. It would be nice to have _Material_ selectable under _Subproject(s) affected_.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-03-17:
Is a separate material property appropriate? It could also be added as a property to Thermal.yml if that's more appropriate

FEA-eng, 2024-03-17:
> Is a separate material property appropriate? It could also be added as a property to Thermal.yml if that's more appropriate

I think that it should be a new property in the Thermal group:

[image]

It could be also named _Thermal expansion reference temperature_ for more clarity bit it would be a rather long name so maybe with some abbreviations? Like _Thermal expansion ref. temp._ ?

PR #13026 (2024-04-01): Material: Add thermal reference temperature
Fixes #13019

Adds the property 'Thermal Expansion Reference Temperature' to the Thermal properties
commit 57c66a96 (2024-04-01): Material: Add thermal reference temperature (#13026)

* Material: Add thermal reference temperature

Fixes #13019

Adds the property 'Thermal Expansion Reference Temperature' to
the Thermal properties

* Fix delegates
closed 2024-04-01 by chennes