FreeCAD/FreeCAD #13022: App::VRMLObject with more than one texture isn't properly displayed when file is reopened

alafr, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I import a VRML (.WRL) file with textures, it initially loads correctly. But when the Freecad file is saved and reopened, only the first texture appears. The other textures are replaced with grey color.

The `Urls `property has different values after reopening the file. I notice that only the first path contains the file name (and its the only texture that reloads correctly), the other paths only have contain the directory path:

- Initial values:
[image]

```
>>> obj.VrmlFile
'C:/Users/antla/AppData/Local/cache/FreeCAD/Cache/FreeCAD_Doc_a5d0c959-ce94-4c04-8104-621937a8cecc_da39a3_7632/Lilac.wrl'
>>> obj.Urls
['C:/Users/antla/Desktop/ngplant/syringa_vulgaris_bark.png', 'C:/Users/antla/Desktop/ngplant/syringa_vulgaris_leaf.png', 'C:/Users/antla/Desktop/ngplant/Lilac-flower.png']
```

- After reopening the file:
[image]

```
>>> obj.VrmlFile
'C:/Users/antla/AppData/Local/cache/FreeCAD/Cache/FreeCAD_Doc_d00fdcec-4834-44f3-a6d1-d268cea9c171_ba1cef_7632/Lilac.wrl'
>>> obj.Urls
['C:/Users/antla/AppData/Local/cache\\FreeCAD\\Cache\\FreeCAD_Doc_d00fdcec-4834-44f3-a6d1-d268cea9c171_ba1cef_7632/Lilac/syringa_vulgaris_bark.png', 'C:/Users/antla/AppData/Local/cache\\FreeCAD\\Cache\\FreeCAD_Doc_d00fdcec-4834-44f3-a6d1-d268cea9c171_ba1cef_7632/Lilac/', 'C:/Users/antla/AppData/Local/cache\\FreeCAD\\Cache\\FreeCAD_Doc_d00fdcec-4834-44f3-a6d1-d268cea9c171_ba1cef_7632/Lilac/']

```
Here is the VRML file:
[ngplant.zip](https://github.com/FreeCAD/FreeCAD/files/14628870/ngplant.zip)

### Full version info

```shell
Tested on these two versions:

OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * BIM
  * Curves 0.6.14
  * Defeaturing 1.2.1

[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * BIM
  * Curves 0.6.14
  * Defeaturing 1.2.1
[/code]
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-18:
That's a regression because it worked with older versions. It worked fine with v0.17 and v0.18 but is broken since v0.19

closed 2024-03-22 by wwmayer

alafr, 2024-03-24:
I tested with Version: 0.22.0dev.36526 (Git) and it works perfectly ! Thanks !

commit fcc7ab4e (2024-03-22): App: fixes #13022

App::VRMLObject with more than one texture isn't properly displayed when file is reopened