FreeCAD/FreeCAD #13023: Error when importing a step file

gue21, 2024-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most of the step models from https://www.multiboard.io/parts-library/remixing show the same behavior
[image]

I wanted to import the files for remixing in a complete empty FCStd file.
gmsh.exe crashes when opening these files too. This may be a problem of my windows installation???

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Austria (en_AT)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I started a discussion in the forum: https://forum.freecad.org/viewtopic.php?p=747665#p747665

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-18:
According to https://forum.freecad.org/viewtopic.php?p=747755#p747755 this is an OCC bug in version 7.6.3

maxwxyz, 2024-03-18:
@wwmayer is this still relevant, are we using this version or a later one for 0,22?

gue21, 2024-03-18:
Wegen des Hinweises auf eine frühere Version habe ich einige Versionen von
gmsh durchprobiert. Sie funktionieren bis zur Version 4.7.1 (occ 7.4.0) für
mich.

danke,
Guenter

wwmayer, 2024-03-18:
The following versions work for me:
* OCC 7.5.1 (default on Ubuntu 22.04)
* OCC 7.6.0
* OCC 7.7.2
* OCC 7.8.0

I guess only a very few OCC releases are affected by this bug.

> is this still relevant, are we using this version or a later one for 0,22?

We cannot rely that every platform uses a version different to 7.6.3. But if we can't find an easy workaround for this OCC version then we should just leave it as is.

wwmayer, 2024-03-18:
I have also created a build with OCC 7.6.3 and it still works.

```
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36460 (Git)
Build type: Unknown
Branch: main
Hash: b4e11db78d8e9cafbda0e4c3f2dc0538e2537919
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: German/Germany (de_DE)
```

maxwxyz, 2024-03-18:
so should we close this issue?

wwmayer, 2024-03-18:
I think we should at least know what is to blame for the issue.

dpasukhi, 2024-09-30:
Hello, for verify from OCCT side we needs STP files.
I would like to not register in that site.
Could you please provide me a some STP files?

luzpaz, 2024-10-15:
> I started a discussion in the forum: https://forum.freecad.org/viewtopic.php?p=747665#p747665

@dpasukhi step examples are available in the forum thread

ElementW, 2026-08-25:
Trying to import [Big Thread - Horizontal, Positive.step.zip](https://github.com/user-attachments/files/31420989/Big.Thread.-.Horizontal.Positive.step.zip) from the forum post fails with `Cannot compute Inventor representation for the shape of Unnamed#Part__Feature: NCollection_Array1::Value
` for all AppImage weeklies from current back to 2025.05.26, and all releases (tested 1.0.0 and above), not sure what @wwmayer is on about unless it broke again between OCC 7.8.0 and 7.8.1.

On a build of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c it fails when built with OCC 7.9.3, but succeeds with 8.1.0.dev1-3d097a0328:

[image]

@maxwxyz I'd say this is resolved upstream and can be closed either now or when we start requiring OCC 8 (or whatever specific version fixed this).

wwmayer, 2026-08-25:
See https://forum.freecad.org/viewtopic.php?p=903340#p903340

maxwxyz, 2026-08-25:
@ElementW we have added the OCC version info to the OCC board to categorize which minimum version fixed an issue:

[image]

stefano2734, 2026-09-12:
Occt 8.0.1 available in libpack 3.5.5 for Linux installation with FreeCAD 26.3 dev

stefano2734, 2026-10-18:
26.3 RC2 available with OCCT 8.0.1.
A successful test can close this issue.