FreeCAD/FreeCAD #13031: saveImage different behaviour in macro

Chocolateandmilkwin, 2024-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

So i have stumbled into this problem a couple of times with different models.
I have a macro that export images of my model from differnt angles, the model consists of an operator panel.
I have inserted a flat body that the operator panel is "mounted" on, so that when i export the images, the rear side components on the panel are not visible.
but on the "Isometric" angle, i sometimes get some bits of the rear components poking through that flat body.
If i try to replicate the same image manually it comes out correctly.
I think the issue is related to the fact that i hide and show the flat body just before i export the image, because if i remove that part, i don't get the issue (I need to call the fitAll method to frame the image correctly, and that does not work when the flat body is shown).
But i still consider it a bug, since the flat body still shows up on some of the picture.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Estimate 0.1.2
[/code]
```

### Subproject(s) affected?

None

### Anything else?

I have included the FreeCad project and a short version of the macro, and two images of what i get when i try it.

To replicate, follow these steps
For the macro change the export path and then run it, you should get an image of the "LMC09-D0121-V001" part with some purple in the background.

For manual, first resize the viewport so it is perfectly square, eg. height pixels is the same as width pixels.
You can use the Gui.activeView().getSize() command to get the viewport pixels.

Then run the saveImage tool and make sure to use the same width and height so it comes out square.

I have used 10000x10000 pixels for both tests, but lower resolutions also has the problem.

[test.zip](https://github.com/FreeCAD/FreeCAD/files/14639369/test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-09:
@Chocolateandmilkwin could you please test this on the latest weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Chocolateandmilkwin, 2024-11-09:
Sadly the issue is still present with the weekly version, below i have attached an image of the output of manual save image on the left and macro on the right.

[image]

Just for good measure heres the version info
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Denmark (en_DK)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

luzpaz, 2025-06-15:
@Chocolateandmilkwin can you recreate on latest 1.1dev ?

Chocolateandmilkwin, 2025-07-20:
I tried with the weekly build [FreeCAD_weekly-builds-42605-conda-Linux-aarch64-py311.AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-42605-conda-Linux-aarch64-py311.AppImage) on linux and the issue sadly persists

Heres the manual image
[image]

Heres the script image
[image]

I am mainly on linux now, but it shouldn't make a difference? The issue is slightly different with the purple bit being slightly skewed.
To my eye it looks like the "camera" is clipping like it does in other 3d software when your viewport gets too close to the

luzpaz, 2025-07-21:
cc @galou any chance you could troubleshoot this ?