FreeCAD/FreeCAD #13034: Cannot open my files anymore. Any workaround?

leoheck, 2024-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After installing `FreeCAD_weekly-builds-36331-2024-03-08-conda-Linux-x86_64-py310.AppImage` I can't open my files again. I have a couple of files that I cannot open anymore.

Now I am using `FreeCAD_weekly-builds-36438-2024-03-15-conda-Linux-x86_64-py310.AppImage` but it is not able to open those files anymore.

Is there anything I can do to be able to open those project files?

### Full version info

```shell
I don't have this AppImage anymore.

FreeCAD_weekly-builds-36331-2024-03-08-conda-Linux-x86_64-py310.AppImage
```
```

### Subproject(s) affected?

None

### Anything else?

I have created this issue before here, 
https://github.com/FreeCAD/FreeCAD-Bundle/issues/214

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-19:
Could you share a file? I cannot reproduce on windows.

leoheck, 2024-03-19:
You won't be able to reproduce it, I guess.

The file was generated with a very specific weekly build I am not sure of the version.

But it should be this `FreeCAD_weekly-builds-36331-2024-03-08-conda-Linux-x86_64-py310.AppImage` or the one before, but I cannot get those AppImages again, to be able to test or reproduce it.

This is what I get when loading one file. 

Do the .FCStd has the version of the Freecad used in the last save? I can try to get it from inside the file if there is this version.

I unfortunately cannot share these files. 

```
➜ freecad case-v0.2.4.FCStd        
Latest FreeCAD /home/lheck/Applications/FreeCAD_weekly-builds-36438-2024-03-15-conda-Linux-x86_64-py310.AppImage
FreeCAD 0.22.0, Libs: 0.22.0devR36438 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42990) [0x7b8bb2042990]
```

leoheck, 2024-03-19:
If I unpack the `case-v0.2.4.FCStd` I can find Freecad version `0.22R36277 (Git)`

```
/tmp/case-v0.2.4.FCStd …
➜ grep -i version *
Document.xml:<?xml version='1.0' encoding='utf-8'?>
Document.xml:<Document SchemaVersion="4" ProgramVersion="0.22R36277 (Git)" FileVersion="1" StringHasher="1">
Document.xml:                <Property name="a2p_Version" type="App::PropertyString" group="importPart" doc="" attr="0" ro="0" hide="0" status="2097152">
Document.xml:                <Property name="a2p_Version" type="App::PropertyString" group="importPart" doc="" attr="0" ro="0" hide="0" status="2097152">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
GuiDocument.xml:<?xml version='1.0' encoding='utf-8'?>
GuiDocument.xml:<Document SchemaVersion="1" HasExpansion="1">
grep: thumbnails: Is a directory
```

If I do the same for my file version `case-v0.2.3.FCStd`, I see a different  Freecad version `0.22R36228 (Git)`

```
/tmp/case-v0.2.3.FCStd …
➜ grep -i version *
Document.xml:<?xml version='1.0' encoding='utf-8'?>
Document.xml:<Document SchemaVersion="4" ProgramVersion="0.22R36228 (Git)" FileVersion="1" StringHasher="1">
Document.xml:                <Property name="a2p_Version" type="App::PropertyString" group="importPart" doc="" attr="0" ro="0" hide="0" status="2097152">
Document.xml:                <Property name="a2p_Version" type="App::PropertyString" group="importPart" doc="" attr="0" ro="0" hide="0" status="2097152">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
Document.xml:                <Property name="_Version" type="App::PropertyInteger" status="83886080">
GuiDocument.xml:<?xml version='1.0' encoding='utf-8'?>
GuiDocument.xml:<Document SchemaVersion="1" HasExpansion="1">
grep: thumbnails: Is a directory
```

leoheck, 2024-03-19:
I remember changing these item of an objects on those files.
[image]

I am wondering if this may be the culprit.

leoheck, 2024-03-19:
I am also wondering if I can get one part (Part Design object from inside of those files)

wwmayer, 2024-03-19:
> FreeCAD_weekly-builds-36438-2024-03-15-conda-Linux-x86_64-py310.AppImage

For more details see: https://github.com/FreeCAD/FreeCAD/pull/12861 and https://forum.freecad.org/viewtopic.php?t=85902

> FreeCAD_weekly-builds-36480-2024-03-18-conda-Linux-x86_64-py311.AppImage

For more details see: https://github.com/FreeCAD/FreeCAD/pull/12035#discussion_r1529031141
This crash is already fixed with https://github.com/FreeCAD/FreeCAD/pull/13033

leoheck, 2024-03-20:
Thanks @wwmayer, for confirming that this `FreeCAD_weekly-builds-36487-2024-03-19-conda-Linux-x86_64-py311.AppImage` new release fixed my issue. Files are opening now.

closed 2024-03-20 by leoheck