FreeCAD/FreeCAD #13036: Crash on launch with latest weekly (36480) on macOS 14.4

JohnOCFII, 2024-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just downloaded the fresh (7 minutes old) weekly.  I get an immediate crash on launch attempt.

Version is 36480.  Mac OS ARM.

Here's the first part of the crash dump:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [10989]
Path:                  /Applications/Other Apps/FreeCAD-36480.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [10983]
User ID:               501

Date/Time:             2024-03-18 18:24:04.6317 -0500
OS Version:            macOS 14.4 (23E214)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       6010C1E9-7187-4F05-9E45-8DD4661AA6DF

Time Awake Since Boot: 100000 seconds
Time Since Wake:       2080 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000004
Exception Codes:       0x0000000000000001, 0x0000000000000004

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [10989]

VM Region Info: 0x4 is not in any region.  Bytes before following region: 4371791868
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104944000-104964000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-36480.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libFreeCADGui.dylib           	       0x10677e024 Gui::MainWindow::_updateActions() + 688
1   libQt5Core.5.15.13.dylib      	       0x10864de84 0x108508000 + 1334916
2   libQt5Core.5.15.13.dylib      	       0x108631094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
3   libFreeCADGui.dylib           	       0x10638e658 Gui::StartupPostProcess::showMainWindow() + 312
4   libFreeCADGui.dylib           	       0x10638cfd8 Gui::StartupPostProcess::execute() + 232
5   libFreeCADGui.dylib           	       0x10629d934 Gui::Application::runApplication() + 1288
6   freecad                       	       0x10494b030 main + 4932
7   dyld                          	       0x193bce0e0 start + 2360

Thread 1:
0   libsystem_pthread.dylib       	       0x193f51d20 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	       0x193f51d20 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	       0x193f51d20 start_wqthread + 0

[... 207 lines cut ...]
__LINKEDIT                       545.6M       66 
__OBJC_RO                         71.7M        2 
__OBJC_RW                         2195K        1 
__TEXT                           411.0M      461 
dyld private memory                272K        2 
mapped file                      185.6M       27 
shared memory                     1392K       16 
===========                     =======  ======= 
TOTAL                              2.8G     2327 

```

### Full version info

```shell
(Can't get that far) Fresh weekly 0.22 - 36480
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-19:
Should be fixed with #13033

closed 2024-03-19 by maxwxyz

JohnOCFII, 2024-03-20:
Confirmed resolved with build: 36487.

```
OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```