FreeCAD/FreeCAD #13040: Disable auto-replacement of input data dimensions.

kaiwas, 2024-03-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the App::PropertyLength parameter has automatic dimension conversion. This is not always convenient.
For example 9999mm + 1mm. automatically replaced by 10 m. And there is no way to force 10000 mm to be displayed. (and continue to work in millimeters).

Switching from millimeters to meters requires the user to pay attention to the input data and decimal places.
If we do not control this, a situation may arise when instead of millimeters or meters we may accidentally end up with kilometers. example in the video.
This requires constant tension from the user.

https://github.com/FreeCAD/FreeCAD/assets/32273833/bc15a48e-e991-4b8a-886b-77b4dd858c58

I think it would be useful to have a setting that allows you to forcefully specify the dimension of the input data (for example, select strictly millimeters in the drop-down list) and disable autocorrect.

[image]

### Full version info

```shell
Any Version
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jwueller, 2024-05-13:
From my cursory understanding, this seems to be caused by expression formatting not actually honoring the user's unit system preferences, and falling back to guessing the exponent based on the value instead, even though it's explicitly set to "mm" in your case (see "Unit system" setting).

So the setting you're asking for is kind of what the "Unit system" setting _should have_ done already, meaning it should show all length units in "mm" everywhere. But for some reason, that just doesn't happen in a lot of places I looked at.

@maxwxyz You can assign this one to me, since it intersects with the expression editor I'm currently working on.

kaiwas, 2024-05-14:
> (see "Unit system" setting)

In my opinion this is a separate setting. To switch systems of quantities. Metric/Imperial/Other.
The metric system can have both millimeters and meters.
In imperial inches and so on.
Let's consider this the first level setting.

The second level setting (in my picture) is already the preferred measurement value. Because We indicate specifically the “scale” in which we work.

My opinion is that these are completely different settings.

jwueller, 2024-05-14:
> In my opinion this is a separate setting.

The current unit system selection does include a specific scale. Here are the current options:

[image]

Either way, changing the unit system selector is somewhat out of scope for this issue, which is mostly caused by the selection not being honored correctly.

I agree with you that it might make sense to take a closer look at how the selector works, but I think it would be a good idea to do discuss that on the forum after we can see how much fixing this issue affects things.