FreeCAD/FreeCAD #13043: FreeCAD.openDocument('filename', hidden=True) segfaults in latest weekly if no document is open

knipknap, 2024-03-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the latest weekly, FreeCAD.openDocument() leads to a segfault.

Steps to reproduce:
- Open FreeCAD (I am using FreeCAD_weekly-builds-36487-2024-03-19-conda-Linux-x86_64-py311.AppImage)
- In the Python console, type `FreeCAD.openDocument('/path/to/a/file', hidden=True)`
- Note that it segfaults.

```
FreeCAD 0.22.0, Libs: 0.22.0devR36487 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.12.3) loaded
1.669602 <asm3.main> init_gui.py(14): no solver backend found
Recent macros : keyboard shortcut(s) 1 disabled because of conflicts with Std_ViewFront respectively.
Hint: In Preferences -> Python -> Macro -> Recent Macros menu -> Keyboard Modifiers this should be Ctrl+Shift+ by default, if this is now blank then you should revert it back to Ctrl+Shift+ by pressing both keys at the same time.
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7efed4fe8520]
#1  0x7efed7e761d9 in Gui::ViewProviderSuppressibleExtension::extensionUpdateData(App::Property const*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0xa9
#2  0x7efed7e65fb7 in Gui::ViewProvider::updateData(App::Property const*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0xd7
#3  0x7efed7e7e851 in Gui::ViewProviderDocumentObject::updateView() from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x101
#4  0x7efed7a2586d in Gui::Document::slotNewObject(App::DocumentObject const&) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x70d
#5  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so(+0x21c0ef) [0x7efed6b9e0ef]
#6  0x7efed6badd6b in App::Document::addObject(char const*, char const*, bool, char const*, bool) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x29b
#7  0x7efed6baf43d in App::Document::readObjects(Base::XMLReader&) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x2cd
#8  0x7efed6bc0491 in App::Document::Restore(Base::XMLReader&) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x2e1
#9  0x7efed6bca04b in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x67b
#10  0x7efed6d9f59d in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x31d
#11  0x7efed6da234e in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x7de
#12  0x7efed6da3fa8 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0xc8
#13  0x7efed6daeaf6 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADApp.so+0x106
#14  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libpython3.11.so.1.0(+0x205a2d) [0x7efed7178a2d]
#15  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x25c) [0x7efed715859c]
#16  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x9977) [0x7efed716d1d7]
#17  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libpython3.11.so.1.0(+0x2aa9ad) [0x7efed721d9ad]
#18  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7efed721d424]
#19  0x7efed7ce9541 in Gui::InteractiveInterpreter::runCode(PyCodeObject*) const from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0xc1
#20  0x7efed7ce9cc0 in Gui::InteractiveInterpreter::runSource(char const*) const from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x60
#21  0x7efed7ce9de8 in Gui::InteractiveInterpreter::push(char const*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0xb8
#22  0x7efed7cea016 in Gui::PythonConsole::runSource(QString const&) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x116
#23  0x7efed7ceb192 in Gui::PythonConsole::keyPressEvent(QKeyEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x492
#24  0x7efed6225a00 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Widgets.so.5+0xf70
#25  0x7efed62ccc6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#26  0x7efed61e63ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#27  0x7efed61edbdf in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Widgets.so.5+0xd1f
#28  0x7efed7a60509 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x99
#29  0x7efed56b9b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Core.so.5+0x108
#30  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca118) [0x7efed6240118]
#31  0x7efed61e63ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#32  0x7efed7a60509 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libFreeCADGui.so+0x99
#33  0x7efed56b9b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Core.so.5+0x108
#34  0x7efed5b422fb in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Gui.so.5+0xdb
#35  /tmp/.mount_FreeCAa8jXy1/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0xbf1b) [0x7efece5c6f1b]
#36  /tmp/.mount_FreeCAa8jXy1/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7efed56ee6f2]
[... 26 lines cut ...]
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.60.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * FreeCAD_assembly3 0.11.0
  * lattice2 1.0.0
  * CycloidGearBox
  * fcgear 1.0.0
  * Assembly4 0.12.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-03-19:
Hi,
will be fixed by https://github.com/FreeCAD/FreeCAD/pull/13032

knipknap, 2024-03-19:
Awesome, Thanks for the quick reaction and the fix!

PR #13032 (2024-04-01): [PD] Fix crash related to SuppressibleExtension
Fix #13001 

- Add a validity test on linked `App::DocumentObject`
- Rename variables for a better readability

EDIT : Fix #13043
closed 2024-04-01 by chennes