FreeCAD/FreeCAD #13044: Problem:  Shapebinder and Subshapebinder have overlapping functionality.  Duplication is confusing

sliptonic, 2024-03-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The overall utility of a (sub)shapebinder is apparent.  But it is confusing to know when a user should use one tool vs another.  
It is possible to select the entire shape and create a subshapebinder so the usefulness of a regular shapebinder isn't clear.
Is this just a legacy thing?  Can old-style shapebinders be eliminated?

### Full version info

```shell
N/A
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-20:
I'm also in favor of only using the green sub shape binder.

FEA-eng, 2024-03-20:
Yes, such redundancy can be really confusing. This wiki section summarizes the differences nicely: https://wiki.freecad.org/PartDesign_SubShapeBinder#PartDesign_SubShapeBinder_vs._PartDesign_ShapeBinder

It would be great to unify those tools without loosing any of their advantages.

macdroid53, 2024-05-01:
Please see my comments in this thread: https://forum.freecad.org/viewtopic.php?p=756621#p756621

nazar-pc, 2024-06-16:
I find sub-object(s) shape binder really confusing. Not only it doesn't allow to select parts of objects, it creates "something" every time I click it, even if nothing is selected at all.
The only thing that I found it helpful for (with my limited experience) is if I want to create a shape binder for the whole sketch, which is still possible to do with regular shape binder, just takes much more time (I described some UX issues with it in https://github.com/FreeCAD/FreeCAD/issues/14705).