FreeCAD/FreeCAD #13046: Problem with using symmetry constraint with rectangles

pami1978, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

with new relase 0.21.2 when you try to use symmetry constraint, it shows you that tere are redundant constraints, but not symmetrize the rectangle, not have this problem with version 21.1

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * A2plus 0.4.64
  * Assembly4 0.50.12
  * boltsfc 2022.11.5
  * dxf-library
  * fasteners 0.5.14
  * frame 0.1.1
  * freecad.gears 1.2.0
  * InventorLoader 1.4.0
  * Marz 0.0.31-alpha
  * sheetmetal 0.4.4
  * symbols_library
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-20:
could you also try this with the weekly build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-03-20:
I guess that you don't have the "Auto remove redundants" option enabled in the Sketcher preferences, I would recommend using it.

closed 2024-03-22 by maxwxyz