FreeCAD/FreeCAD #13048: Black 3D view on MacOS

markuspalme, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On MacOS the 3D view appears as a black window:

[image]

Switching to Software OpenGL makes it work.

Same behaviour with the latest stable and the weekly dev build.

### Full version info

```shell
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

markuspalme, 2024-03-20:
I noticed this output in the log:

```
09:31:14  (qt.qpa.openglcontext) Could not create NSOpenGLContext with shared context, falling back to unshared context.
09:31:14  (qt.qpa.openglcontext) Failed to create NSOpenGLContext
09:31:14  QOpenGLWidget: Failed to create context
09:31:14  (qt.qpa.openglcontext) Could not create NSOpenGLContext with shared context, falling back to unshared context.
09:31:14  (qt.qpa.openglcontext) Failed to create NSOpenGLContext
09:31:14  QOpenGLWidget: Failed to create context
09:31:14  No valid GL context found!
09:31:14  No valid GL context found!
```

markuspalme, 2024-03-21:
After a reboot of the system, it's working again. Not sure if I should close this bug.

luzpaz, 2024-09-08:
@markuspalme have you been able to reproduce ?

closed 2024-09-09 by markuspalme