FreeCAD/FreeCAD #13055: PartDesign_Plane display error if clicked with a base plane selected, but allows base plane selection in dialog.

macdroid53, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With a base plane selected, PartDesign_Plane displays error when clicked: "There are no attachment modes that fit the selected objects. Select something else." Then opens the usual dialog and allow the user to select the plane that was selected to begin with. The DP is created when the Ok button is clicked.
This is the behavior in 36491, in 36438 it caused a segfault.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36491 (Git)
Build type: Release
Branch: main
Hash: e4edf3afa6eae8068a9ea0273cb003b32b773040
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * OpenDark 2023.12.17
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Curves 0.6.29
  * ArchTextures
  * Nodes 0.1.36
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.3
  * Silk 0.1.5
  * Freecad-Built-in-themes-beta 1.2.2
  * kicadStepUpMod 11.1.0
  * DynamicData 2.60.0
  * workfeature-macro
  * fasteners 0.5.14
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.8
```

### Subproject(s) affected?

None

### Anything else?

This behavior is seen with a new file with a PD Body. Select a base plane, then click, PartDesign_Plane.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-21:
I can confirm this behaviour and it's a regression compared to v0.21 where it works correctly.

wwmayer, 2024-03-21:
When I restore Attacher.cpp and Attacher.h of 7945bf686e5d and apply the changes of 12861 then the crash is fixed and attaching to the datum plane works as expected.

So, the PR to blame is https://github.com/FreeCAD/FreeCAD/pull/12885

bgbsww, 2024-03-21:
Thanks for this.  There is clearly an issue with references, and equally clear is that there is still some code that doesn't match correctly after the transfer.  I'm actively going through this now that I see the problem.

PR #13082 (2024-03-21): Toponaming/Part: Move missed attacher code into place
Fix #13055.   Passes #13078.  Also passes all other c++ and python tests.