FreeCAD/FreeCAD #13057: Assembly: Treeview collapses while adding a Sketch

FreeCutter, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Repro:**
1. Adding a new Assembly (WB Assembly: Assembly > Create Assembly)
2. Adding a Body to active Assembly (WB Part Design: Part Design > Create body)
3. Adding Sketch to active Body (WB Part Design: Sketch > Create sketch).
4. Selecting e.g. XY-plane from Task 'Select feature' and click OK
 
**Expected results:**
- Treeview remains expanded
   [image]
 

**Actual result:**
-  Treeveiw gets collapsed
   [image]

Comment: this only happens on the first run. Once another Sketch is added (repeating step 3. onwards) the Treeview keeps expanted.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * fasteners 0.5.14
  * SelectorToolbar
```

### Subproject(s) affected?

Assembly

### Anything else?

[Test.zip](https://github.com/FreeCAD/FreeCAD/files/14668142/Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-21:
So this happens because opening a sketch is unsetting the assembly edit mode. Which is unsetting the assembly as the active App::Part.

Whenever a App::Part is unset as the active App::Part, it collapse.
You can try to create a App::Part, then create a second one. The first one will collapse as it loose the activation state.

The same way double clicking several time a App::Part toggles it's activation state, making it expand/collapse.

So I am not sure what to do. Do you guys think that unsetting the 'activated' state should not collapse the assembly? ie double clicking on assembly object will not collapse them.
I am not even sure I can customize this behavior for AssemblyObject or if it has to be the same for all App::Part.

FreeCutter, 2024-03-22:
Thanks for your feedback.
I run the following two variantes A and B to illustrate what's going on:
**Var. A:**
- Opening Test.FCStd 
  Part_Assembly is: not activated; not in assembly-edit-mode; expanded; displayed: nothing

- **One**-click on Assembly object in the treeview
  Part_Assembly: not activated; not in assembly-edit-mode; expanded; displayed: blue background, not bold letters)

- Righ-click on Assembly object in the treeview to opening context menue and click on 'Active object'
  Part_Assembly: activated; not in assembly-edit-mode; expanded; displayed: blue and bold)

- Adding a Body object via WB Part Design
   Part_Assembly: activated; not in assembly-edit-mode; expanded; displayed: blue and bold
   Body001 is activated

- Creating sketch for activated Body001 for e.g. XY plane getting into sketcher mode
  Part_Assembly: activated; not in assembly-edit-mode; expanded; displayed: blue and bold
  Body001 is activated

**Var. B:**
- Opening Test.FCStd 
  Part_Assembly is: not activated; not in assembly-edit-mode; expanded; displayed: nothing)

- **Doube**-click on Assembly object in the treeview
  Part_Assembly: activated; in assembly-edit-mode; expanded; displayed: yellow background and bold letters

- Adding a Body object via WB Part Design
   Part_Assembly: activated; in assembly-edit-mode; expanded; displayed: yellow and bold
   Body001 is activated

- Creating sketch for activated Body001 for e.g. XY plane getting into sketcher mode
  Part_Assembly: not activated; not in assembly-edit-mode; collapsed; displayed: nothing
  This case is the issue

To avoide this issue I see this possibility:
- When Body is added to a Part_Assembly being in edit-mode then the Part_Assembly is put into "normal" Part status appropiate for this situation which would be Part_Assembly behaving like in Var. A. 

Comment: treatment/management of the 'assembly-edit-mode' needs to be checked regarding:
  - how to enter into assembly-edit-mode:
    E.g.: in case Var. A (Part_Assembly: just normal active, i.e. not in assembly-edit-mode': double click to enter into assembly-edit-mode does not work (?) 
  - how to leave assembly-edit-mode actively by user action:
    Option: Via context menu, i.e. toggeling 'Active object'. But that's deactivating and collapsing the whole Part_Assembly which is considered to be uncomfortable.
Maybe this 'comment' needs to be moved to another issue in order to be better processed?

PaddleStroke, 2025-11-27:
So now this problem is less a problem, because when you finish editing the sketch, the assembly gets back into edit, hence reopening itself.

But still it's not ideal.

One way I think we can fix this, is that when sketch starts edit, it can make sure to open the assembly