FreeCAD/FreeCAD #13061: TechDraw: Double clicking on a page does not switch workbench to TechDraw

PaddleStroke, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In sketcher double clicking on a sketch changes the active workbench to sketcher.
Similarly double clicking on an assembly switch active workbench to assembly.

For consistency we could have the same behavior when double clicking on a page, switch to techdraw.
Thoughts on this?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-03-23:
Sketcher is the only place this happens (at least out of the pieces of FC that I use).  It doesn't happen for TD, Spreadsheet or Draft.  So I'm not sure this advances the cause of consistency.

I like the idea, though.  I've wasted lots of time looking at a spreadsheet, thinking I'm in SpreadsheetWB, pressing keys and not getting the expected response.

PaddleStroke, 2024-03-25:
Perhaps then it would be good to implement that in spreadsheet as well. In draft I'm not sure in which context that could happen. It's not like there is a main 'draft object'.

PaddleStroke, 2024-03-25:
https://github.com/FreeCAD/FreeCAD/issues/13136

PR #13063 (2024-03-25): TechDraw: Fix drag&drop of views. Add auto-switch to techdraw on double click.
This PR address 3 low hanging fruits in techdraw : 
- Fixes #13061
In sketcher double clicking on a sketch changes the active workbench to sketcher.
Similarly double clicking on an assembly switch to assembly.
For consistency, I offer to implement the same behavior when double clicking on a page, switch to techdraw.

- Fixes #12375 :
Techdraw has a 'move view' tool that is pointless. Moving views should be done by drag and drop. It actually was supported before but got broken with the introduction of `canDropObjectE [... cut]
commit fae245be (2024-03-25): TechDraw: double clicking page switch to techdraw wb. Fixes #13061
closed 2024-03-25 by WandererFan