FreeCAD/FreeCAD #13067: Deleting a centerline followed by undo → segfault

mpfusion, 2024-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Deleting a centerline in TechDraw followed by undo leads to a segfault.

Steps to reproduce
---------------------------
- open the file 
[add-centerline.tar.gz](https://github.com/FreeCAD/FreeCAD/files/14671687/add-centerline.tar.gz)
- select the dotted centerline
-  \<delete>
- \<ctrl>-z
- → segfault

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c510) [0x7f520505a510]
#1  0x7f51c0d78211 in TechDraw::DrawUtil::getGeomTypeFromName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x11
#2  0x7f51c0ec1b9a in TechDraw::CenterLine::calcEndPoints(TechDraw::DrawViewPart const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, int, double, double, double, double) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x10a
#3  0x7f51c0ec4611 in TechDraw::CenterLine::scaledAndRotatedGeometry(TechDraw::DrawViewPart*) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x2b1
#4  0x7f51c0eeca70 in TechDraw::CosmeticExtension::addCenterLinesToGeom() from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x180
#5  0x7f51c0dfc7b5 in TechDraw::DrawViewPart::postFaceExtractionTasks() from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x25
#6  0x7f51c0df6218 in TechDraw::DrawViewPart::onFacesFinished() from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x128
#7  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x3062b2) [0x7f52059062b2]
#8  0x7f52056e5035 in QFutureWatcherBase::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x145
#9  0x7f5206562f32 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x82
#10  0x7f520861f278 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#11  0x7f52058cc748 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#12  0x7f52058cfe51 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x191
#13  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x328697) [0x7f5205928697]
#14  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x571f4) [0x7f5203b111f4]
#15  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x5a317) [0x7f5203b14317]
#16  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7f5203b14930]
#17  0x7f5205927d4a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x6a
#18  0x7f52058cb0fb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#19  0x7f52058d38a4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#20  0x7f52085512e9 in Gui::Application::runApplication() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x9d9
#21  freecad(+0x79e1) [0x559f28a239e1]
#22  /lib/x86_64-linux-gnu/libc.so.6(+0x276ca) [0x7f52050456ca]
#23  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f5205045785]
#24  freecad(+0x7d21) [0x559f28a23d21]
```

https://forum.freecad.org/viewtopic.php?t=86266

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36458 (Git)
Build type: Release
Branch: main
Hash: 934395913a6f9298f7b56bfe3b58c163f7011186
Python 3.11.8, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.3
  * Assembly4 0.50.9
  * BOLTSFC
  * kicadStepUpMod 11.1.0
  * fasteners 0.5.14
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mpfusion, 2024-03-21:
Still present in master `0.22.0devR36500` 161b0fb6ee9418473bc19b36976d3f06c4ca098c.

WandererFan, 2024-03-23:
Jotting my notes here instead of on a scrap of paper...

To reproduce
- select a cosmetic edge (or center line or cosmetic vertex) in a DrawViewPart
- press delete
- press undo
- crash

What happens:
- ViewProviderViewPart::onDelete removes the edge and returns *false* so StdDelete will 
  not delete the DrawViewPart
- the delete transaction remains on the undo stack (and still points at the DVP  and the cosmetic edge
- pressing undo attempts to undo the delete transaction
    - this causes a background/hidden update to the cosmetic edge list property with an
      invalid cosmetic edge pointer.  This invalid cosmetic edge pointer causes a seg fault later.
    - should StdDelete be adding a rejected deletion to the the undo stack?  Doesn't
      seem correct.
- the delete transaction is locked an can not be aborted.

The delete key is trapped by shortcut handling before we can decide to process it or not in 
QGVPage/QGVNavStyle.

We need to add a filter on Edges & Vertices with "if (event->type() == Qt:ShortcutOverride) { " to allow the keystroke to be passed to the QGVNavxx keystroke handler as a regular key press event.

Then we add a "delete key bypass" that checks the selection for cosmetic items.  If there is a selected cosmetic item, we delete it without involving Std_Delete.  If there is no cosmetic item in the selection, we forward the event for 

It would be nice to handle the key press in QGIEdge, but key press events are not getting to QGIEdge (or QGSPage).   Focus issue perhaps?

The only downside is that it is not possible to undo a deletion of a cosmetic item, but that is the current situation, so this solution doesn't make anything worse and prevents the crash.

mpfusion, 2024-04-04:
I can confirm that it's no longer crashing now. Undo still doesn't work, but that's another issue and not related to the reported segfault.

luzpaz, 2024-04-04:
Thanks for the follow-up. Is there a ticket linking to the pre-existing undo bug ?

closed 2024-04-04 by luzpaz

mpfusion, 2024-04-04:
I have found no existing report regarding that issue and I also haven't  created a report, yet.

There are several actions that can't be undone. E.g. cosmetic vertices, adding hole/shaft fits, … and probably a few more. Feel free to create one, otherwise I'll do that at some point whenever I find the time.

luzpaz, 2024-04-04:
@mpfusion please do so when you have the time :pray:  
Then we can include them to the list of the open undo/redo bugs (https://github.com/FreeCAD/FreeCAD/labels/Undo%2FRedo)

mpfusion, 2024-04-24:
@luzpaz Done: https://github.com/FreeCAD/FreeCAD/issues/13618