FreeCAD/FreeCAD #13069: OpenSCAD import duplicates a scaled sphere

novalis, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have an OpenSCAD file which looks like this:

`translate([0, 0, 29]) scale([1, 1, 0.5]) sphere(15); `

(That's it -- that's the whole repro!)

I expect to see one scaled sphere.  Instead, I see one scaled sphere and one unscaled one.
[image]

I'm using the flatpak version of FreeCAD, since that one seems to support OpenSCAD.

And I see the following in the output:
```
WARNING: Token 'DOT' defined, but not used
WARNING: Token 'WORD' defined, but not used
WARNING: There are 2 unused tokens
End processing CSG file
```

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-21:
Could you test this in the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

novalis, 2024-03-23:
It looks like that weekly build doesn't do OpenSCAD import for some reason.  Do you have a build that does?

luzpaz, 2024-08-24:
@KeithSloan can you reproduce this ?

KeithSloan, 2024-08-27:
I get the same result with the standard importer but the Alternate Importer works just file.
[image]
[DoubleSphere.scad.txt](https://github.com/user-attachments/files/16765926/DoubleSphere.scad.txt)
[DoubleSphereAltImporter.FCStd.txt](https://github.com/user-attachments/files/16765929/DoubleSphereAltImporter.FCStd.txt)

luzpaz, 2024-08-27:
@novalis you can use the Alternate OpenSCAD Importer through the [Addon Manager](https://wiki.freecad.org/Std_AddonMgr)

novalis, 2024-08-30:
I just tried installing said add-on.  It apparently installed successfully, and FreeCAD restarted.  But when I open my repro file from File > Open, I am not prompted to use the alternate importer, and I still see the behavior of the default importer.

Am I missing a step?

KeithSloan, 2024-08-30:
If you do a FreeCAD  | About freecad

Copy clipboard

And Past somewhere you should get a list of installed Addon as well as FC info

OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportExport_3DM 0.1.0
  * DynamicData 2.62.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * SWPart
  * toSketch 1.0.1
  * Curves 0.6.42

Then if you do a 
File | Open
and choose a file with a scad or csg extension you should get a window.
For scad it should look like

[image]

I can recollect that sometimes dialog windows can get lost/off screen but cannot remember how one resolves this.

novalis, 2024-09-01:
Ok, I just tried it again, and this time for some reason I did get the importer selection dialog.  Indeed, the new importer works great.  Thanks!

Any chance of making the alternate importer into the default, since it seems to work better?