FreeCAD/FreeCAD #13070: TechDraw: Projection Group use current camera rotation by default

PaddleStroke, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

One thing that struck me as very weird in techdraw was that proj group uses current camera rotation as the 'front view'. I think it is unintuitive and confusing.

I would suggest that projection group uses the object front view as it's default front-view. If there are usecases to want the camera to be the default front view, then I suggest adding a checkbox 'Use camera as front view' unchecked by default.

Thoughts on that?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-22:
Maybe not use the default front view as standard, as in most industries it's common to design parts with a specific direction, especially for the use in assemblies (e.g. +Z up or -X being the "front"). What about a dialog or dropdown to either choose the current view or one of the main views?

PaddleStroke, 2024-03-25:
I think in most cases, the 3dview-front makes more sense for a projection group rather than current camera orientation.

Currently the user wants to make a projection group of a nice house : 
[image]

Here is what he's 99.99% of the time expecting to get : 
[image]

But no, here is what he gets : 
[image]

I don't say we shouldn't offer the possibility to use current camera orientation, just that it should not be the default.

What about having a combobox offering modes : 
- 3DView front view
- Camera orientation
- Manual orientation

That would tackle #5821 at the same time.

closed 2024-04-24 by WandererFan