FreeCAD/FreeCAD #13071: TechDraw: Radius/Diameter of Ellipses are approximates

PaddleStroke, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
When you try to apply a radius or a diameter to an ellipse it gives you an approximate value.
I would say that ellipses in techdraw are almost allways circles viewed from an angle. So it would make more sense to show by default the major radius rather than an approximation.
I don't think there is any usecase in showing an approximation rather than the major radius.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-21:
I guess we need more discussion to this topic.

WandererFan, 2024-06-10:
The major radius of the ellipse is not necessarily the radius of the original circle.

closed 2024-06-10 by WandererFan