FreeCAD/FreeCAD #13072: TechDraw: Cosmetic Eraser should not be a tool.

PaddleStroke, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cosmetic Eraser should not be a tool.

Select the element, press delete.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-03-21:
This is already implemented: https://github.com/FreeCAD/FreeCAD/pull/10695

PaddleStroke, 2024-03-21:
Yes I have seen delete works.
My point is that the eraser tool should be removed from the toolbar.

FEA-eng, 2024-03-21:
> Yes I have seen delete works. My point is that the eraser tool should be removed from the toolbar.

Right, it doesn't make sense to keep it anymore.

closed 2024-03-26 by WandererFan