FreeCAD/FreeCAD #13080: STL exports complex object as separate parts in the STL file in version 0.21.2

nhc155, 2024-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have complex models with multiple bodies. When exported to STL (in version 0.21.2) the single STL file has separated the bodies into multiple objects, instead of a single whole. Not sure if this was the behaviour from when I first installed 0.21.2, or if it just started happening today.

I have reverted to 0.21.1 and it works as expected here.

The attached pictures are screen caps of the STL preview. Good from 0.21.1, bad from 0.21.2.
[image]
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * AirPlaneDesign 0.4.0
  * CurvedShapes 1.0.5
  * Curves 0.6.21
  * Plot 2022.4.17
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-21:
It's fixed with https://github.com/FreeCAD/FreeCAD/commit/3aa6e37bb2c76e07ac68f42dfecf2a92b4c32aa9

closed 2024-03-21 by wwmayer

nhc155, 2024-03-22:
Thank you.

 
 

It's fixed with 3aa6e37