FreeCAD/FreeCAD #13089: Loss of already entered data due to mouse movement.

kaiwas, 2024-03-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

the counters in the sketcher have a problem that is present in all input fields in the program. With the value entered, if we move the mouse in the 3D window, the data is lost. Can this be somehow solved?
If the user has already entered data, he should not lose it due to accidental mouse movement.

You have to hold your breath and stop your heartbeat to prevent this from happening.

https://github.com/FreeCAD/FreeCAD/assets/32273833/8c30d31b-a114-48b9-948e-fb1f38d64348

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Anything else?

This problem occurs everywhere where a 3D window is involved.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-03-22:
You need to press Tab or Enter to confirm your input.

kaiwas, 2024-03-22:
I understand it.
But before pressing TAB, accidental mouse movement may occur.
There are times when you have to think about the entered value or whatever (it takes some time to enter).
Those. There may be a pause between entering the first character and the last one, as well as pressing the tab key itself.
Just experiment a few times without letting go of the mouse and you may realize that the problem may be quite serious. She's everywhere!

mpfusion, 2024-03-22:
Another idea is to use some kind of hysteresis, so the user has to move the cursor at least by *x* amount away to trigger “freeing” the tool again and discarding the entered value.

JohnOCFII, 2024-03-23:
I agree this is a frustrating issue.  I know of no other program that, after activating data entry into a text field, allows mouse movements to change that value.  Working with a laptop with a touch pad, it is VERY common to brush your hands over the touchpad, which changes values.  Super annoying.  I think there were some recent fixes done in the `Draft` workbench to address this in that WB, but this is a problem across many parts of FreeCAD.

benj5378, 2024-10-04:
I might be able to look at this. @maxwxyz, can you assign me, such that I don't forget

maxwxyz, 2025-03-09:
any news on this @benj5378 ?

benj5378, 2025-03-12:
I had a brief look. The locking feature is already implemented for when entering has finished (after pressing tab for moving to next input).

It just need to be extended to also cover inputs currently being entered. I had a hard time finding the code location, and didn't get further

closed 2025-06-02 by kadet1090