FreeCAD/FreeCAD #13104: Raspberry Pi half black screen

n9szf, 2024-03-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When tring to run Freecad version 0.19 on a Raspberry Pi , getting a black screen that runs diagonal across the screen.
[image]

### Full version info

```shell
0.19

2022/09/1400:07:06
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-03-23:
What is your OS? Have you installed _FreeCAD_ from repos or run _AppImage_/_Flatpak_? Which _Raspberry Pi_? What graphic driver do you use? Do you use _X11_ or _Wayland_? Does other software using _OpenGL_ work? What is your screen resolution?

Does _FreeCAD_ actually work? Can you provide Help → About info?

The blue field you see on the screenshot comes from non-working _OpenGL **1.0**_ (yes, that's right, **1.0**). Can you install _Mesa_ and launch _FreeCAD_ with it?

> 2022/09/1400:07:06

What's that?

maxwxyz, 2024-03-23:
could you try the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
0.19 is some time ago.

maxwxyz, 2024-06-15:
I'll close this for no further feedback

closed 2024-06-15 by maxwxyz

jose1711, 2024-11-27:
Same issue with FreeCAD 0.20.2.

> What is your OS?

Raspbian GNU/Linux 12

> Have you installed FreeCAD from repos or run AppImage/Flatpak?

from repos
```
freecad 0.20.2+dfsg1-4+rpi1
```

> Which Raspberry Pi?

Raspberry Pi 5, 8GB model

> What graphic driver do you use? Do you use X11 or Wayland?

Wayland

> Does other software using OpenGL work?

yes (`glxgears`)

> What is your screen resolution?

1280x1024

jose1711, 2024-11-27:
FreeCAD shows initial screen without issues but when I click "Create new..." the following appears inside terminal:
```
Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
```
and part of the screen is covered with black area. This makes it impossible to use the program.

xtemp09, 2024-11-27:
@jose1711, could you launch _FreeCAD_ with the following command:

```ruby
$ QT_QPA_PLATFORM=xcb freecad
```

?

jose1711, 2024-11-28:
> @jose1711, could you launch _FreeCAD_ with the following command:
> 
> ```ruby
> $ QT_QPA_PLATFORM=xcb freecad
> ```
> 
> ?

No change.

jose1711, 2024-11-28:
It's failing for me with this:
```sh
$ ./FreeCAD_weekly-builds-39315-conda-Linux-aarch64-py311.AppImage  
/tmp/.mount_FreeCAeiEipO/AppRun: line 31: /tmp/.mount_FreeCAeiEipO/usr/bin/freecad: cannot execute: required file not found
```

Was unable to find a solution online.

luzpaz, 2024-11-28:
> Same issue with FreeCAD 0.20.2.

What about latest stable ? https://blog.freecad.org/2024/11/19/freecad-version-1-0-released/

jose1711, 2024-11-28:
I guess the issue with AppImage comes from the fact that Raspbian is using 32bit userland whereas AppImage is 64bit.