FreeCAD/FreeCAD #13106: feature request: add G73 peck drilling variant to grbl post processor

wolfgangr, 2024-03-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current version of grbl postprocessor implements G81, G82 and G83 versions of canned drill cycles.

There is also a G73 version of peck drilling. It may be generated by the Path tool in certain circumstances (may be shallow holes and peck drill enabled?), But if so, it is not correctly handled by the preprocessor. (reproducible testing still TBD).

Here is a good explanation of the issue:
[G73 vs G83](https://www.machinistguides.com/g73-code/#G73_vs_G83)

I think it may be implemented on top of the already implemented canned cycles.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 10 (buster) (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.13
  * slic3r-tools
  * Pyramids-and-Polyhedrons
  * A2plus 0.4.64
  * Assembly3 0.12.2
  * Assembly4 0.50.9
  * FeedsAndSpeeds 0.5.0
  * dodo 1.0.1
  * frame 0.1.1
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * MOOC 2022.4.21
  * Help 1.0.3
  * ose-piping
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.2
  * ThreadProfile 1.89.0
  * SteelColumn
  * Estimate 0.1.2
  * DynamicData 2.60.0
  * parts_library
  * btl 0.9.9
  * AnimationFreeCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.8
[/code]
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wolfgangr, 2024-03-23:
> (reproducible testing still TBD).

to prepare targeted testing, lets do a "rtfSource" first: 
github serach `"repo:FreeCAD/FreeCAD path:/^src\/Mod\/CAM\// G73"`

Let's look at some outstanding results:

https://github.com/FreeCAD/FreeCAD/blob/aac48eb2f9ff485af6e7f8b7b7d4977f70fb7789/src/Mod/CAM/GCode-description.md?plain=1#L19-L26

So I think G73 is considered essential part of CAM/Path output- So It's a good idea to handle it.
We also see that G98 and G99 related to canned cycles. Grepping through postprocessors, we quite often find them in "suppressed commands" lists. Is this a good idea? Move XY when Z is at a different level than it should? grbls postpocessor has more hits than any other on G98 / G99. Let's Just keep this in mind.

https://github.com/FreeCAD/FreeCAD/blob/aac48eb2f9ff485af6e7f8b7b7d4977f70fb7789/src/Mod/CAM/Path/Post/UtilsArguments.py#L311-L312

This just adds to the finding that G73 is essential part of canned drill cyles. There are matches in other utils, tests, simulator, ... as well, supporting this, without any obvious relevance for the moment.

For the moment, I consider this as the **essence**:

https://github.com/FreeCAD/FreeCAD/blob/aac48eb2f9ff485af6e7f8b7b7d4977f70fb7789/src/Mod/CAM/Path/Base/Generator/drill.py#L40-L44

   ...

https://github.com/FreeCAD/FreeCAD/blob/aac48eb2f9ff485af6e7f8b7b7d4977f70fb7789/src/Mod/CAM/Path/Base/Generator/drill.py#L53-L60

There is a "Chip Break" Tickmark in the Drilling Operation GUI. It is greyed out until "Peck" is enabled. "Dwell" is, the other way round, disabled as soon as "Peck" is ticked. Looks like a **test plan.**...

We also find the `linuxcnc` link in the last line. This contains a detailled description of the movement of G73 as it is supposed to be implemented.

wolfgangr, 2024-03-23:
_to prepare a test setup, we have to figure out how our settings are supposed to be translated, and compare that with the final result. We stumble over a known mess of OpWhatever variables, Setup sheet, Values complaining if "mm" dimensions are given, just to complain about missing units afterwards ... That's not our job at hand, here, however. We hope this things may change to the positive in some foreseeable future and keep it out of our ways:_

So we set as much as possible in the Operation's Property Editor value sheet:

- clearance height = 10  mm
- final depth = -6 mm
- safe height = 2 mm
- start depth = 1 mm

There is a "Retract Mode" in the proeprty editor, where we can select "G98" or "G99". remember:

    G98 - retract to the position that axis was in just before this series of one or more contiguous canned cycles was started.
    G99 - retract to the position specified by the R word of the canned cycle.

This setting is ignored, however, and the internal gcode previewer always comes up with a G98:

```
(Drilling)
(Begin Drilling)
G0 F8.333333 Z10.000000
G90
G98
G0 F33.333333 X10.000000 Y-1.000000
G81 F1.666667 R-1.000000 X10.000000 Y-1.000000 Z-6.000000
G0 F33.333333 X24.000000 Y10.000000
G81 F1.666667 R-1.000000 X24.000000 Y10.000000 Z-6.000000
```

However, gcode-parser ignores this as well and implements the behaviour of G99:

```
(Drilling)
(Begin Drilling)
G0 Z10.000 F500.000
G90
( G98 )
G0 X10.000 Y-1.000 F2000.000
(G81 X10.000 Y-1.000 Z-6.000 F100.000 R-1.000)
G0 X10.000 Y-1.000
G0 Z-1.000 F100.00
G1 Z-6.000 F100.00
G0 Z-1.000
G0 X24.000 Y10.000 F2000.000
(G81 X24.000 Y10.000 Z-6.000 F100.000 R-1.000)
G0 X24.000 Y10.000
```

**Correct G98** behaviour would be to **revert** after each drill cycle, **before the XY-move, to Z10** as set in the first line, right before G98. What it **actually** produces is **G99 behaviour - retract to R-value.** In the example, where I intended to have R just below the surface to keep a pecking drill in the hole, this would result in a **tool crash** plouging through the stock at jog speed. Glad to say that both the CAM preview (errously, imho) and Openbuilcontrol (simulating `gcode_post.py` output) warn us. So we continue to keep an eye on the G98/99 issue and set "Retract Height" to SafeHeight=2mm for the next test steps.

wolfgangr, 2024-03-23:
next step: test G82 - "dwell cycle"

We use the Operations GUI to tick "Dwell" and set a value of P=789 (milliseconds) - just to identify. 

Internal gcode looks fine - G82 is correctly applied and P=789 forwarded:
```
G0 F33.333333 X10.000000 Y-1.000000
G82 F1.666667 P789.000000 R2.000000 X10.000000 Y-1.000000 Z-6.000000
```

gcode_post.py makes a good job here, I think:
```
(Begin Drilling)
G0 Z10.000 F500.000
G90
( G98 )
G0 X10.000 Y-1.000 F2000.000
(G82 X10.000 Y-1.000 Z-6.000 F100.000 R2.000 P789.0)
G0 X10.000 Y-1.000
G0 Z2.000 F100.00
G1 Z-6.000 F100.00
G4 P789.0
G0 Z2.000
G0 X24.000 Y10.000 F2000.000
(G82 X24.000 Y10.000 Z-6.000 F100.000 R2.000 P789.0)
```
neither CAM preview nor openbuildscontrol seem to correctly implement the G4 pause command. Might test it on a real machine some time...

wolfgangr, 2024-03-24:
next test: pecking

untick `'[ ] Dwell'`, tick `'[X] Peck'` and set Peck Depth to a small positive dimension.

internal gcode looks fine - switched correctly over to G83
```
(Drilling)
(Begin Drilling)
G0 F8.333333 Z10.000000
G90
G98
G0 F33.333333 X10.000000 Y-1.000000
G83 F1.666667 Q1.000000 R2.000000 X10.000000 Y-1.000000 Z-6.000000
```

and parsed output .....
```
(Begin operation: Drilling)
(Path: Drilling)
(Drilling)
(Begin Drilling)
G0 Z10.000 F500.000
G90
( G98 )
G0 X10.000 Y-1.000 F2000.000
(G83 X10.000 Y-1.000 Z-6.000 F100.000 Q1.000 R2.000)
G0 X10.000 Y-1.000
G0 Z2.000 F100.00
G1 Z1.000 F100.00
G0 Z2.000
G0 Z1.050
G1 Z0.000 F100.00
G0 Z2.000
G0 Z0.050
G1 Z-1.000 F100.00
G0 Z2.000
G0 Z-0.950
G1 Z-2.000 F100.00
G0 Z2.000
G0 Z-1.950
G1 Z-3.000 F100.00
G0 Z2.000
G0 Z-2.950
G1 Z-4.000 F100.00
G0 Z2.000
G0 Z-3.950
G1 Z-5.000 F100.00
G0 Z2.000
G0 Z-4.950
G1 Z-6.000 F100.00
G0 Z10.000
G0 X24.000 Y10.000 F2000.000
(G83 X24.000 Y10.000 Z-6.000 F100.000 Q1.000 R2.000)
```
.... basically looks ok, but has some quirks:

- for deep holes, it would be preferrable to keep the drill below surface. But this only can work with a proper G98 implementation, as shown above
- The retraction G0 is rendered without F value. Since positioning the drill down is rendered as G0 as well, but with an F value, this is subsequently used. Pecking gets really slow this way.

This is related to a known issue already solved these days.

wolfgangr, 2024-03-24:
next test: set `'[X] chip break'` in teh GUI

internal gcode correctly switches to G73
```
(Drilling)
(Begin Drilling)
G0 F8.333333 Z10.000000
G90
G98
G0 F33.333333 X10.000000 Y-1.000000
G73 F1.666667 Q1.000000 R2.000000 X10.000000 Y-1.000000 Z-6.000000
G0 F33.333333 X24.000000 Y10.000000
G73 F1.666667 Q1.000000 R2.000000 X24.000000 Y10.000000 Z-6.000000
```

`gcode_post.py` feeds it thrugh, without any change.
I know my mill to throw an error:22 "not implemented" on that.
```
(Begin Drilling)
G0 Z10.000 F500.000
G90
( G98 )
G0 X10.000 Y-1.000 F2000.000
G73 X10.000 Y-1.000 Z-6.000 F100.000 Q1.000 R2.000

wolfgangr, 2024-03-24:
**Summary** result of above tests so far - **work to be done:**

- **clone G83** code and transfer it **to G73** behaviour
- **implement G98** in a way that not breaks backwards compatibility. Think of config disabled by default
- add **proper speed** setting to G83 output (may be before cloning....)
- write some **instruction** how the gui and configuration of drill operations maps to canned drill cycles

wolfgangr, 2024-03-24:
**G98 implementation** plan:

The target behaviour as of here is quite complicated
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g98-g99

We restrict to the case that a "contiguous canned cycle" is generated by a single Path Jobs operation and that it is started by a G98. so it might suffice to

- store current Z at G98 call as **`GLOBAL G98_Z_RETURN`**
- keep a flag indicating G98 behaviour (off by default, cleared by G99)
- at the end of each drill cycle, we move to **`( G98_FLAG`** ` ? max(R-level, G98_Z_RETURN)  : R-level)`
- we assume absolute coordinates

**config options:**
translate_G98
- off (default G99 behaviour)
- fixed 0.000 (set Zlevel by config and ignore G98 / G99)
- simple ( as described here)
- full (not implemented)

wolfgangr, 2024-03-24:
ah, it's alrady provided for:

https://github.com/FreeCAD/FreeCAD/blob/d5b90e50af6a758af748179f289bb8f09e357266/src/Mod/CAM/Path/Post/scripts/grbl_post.py#L636-L637

wolfgangr, 2024-03-24:
roughly tested - looks go0d:
```
631,632c631,632
<     strG0_Initial_Z=("G0 Z" + format(float(CURRENT_Z.getValueAs(UNIT_FORMAT)), strFormat) + "\n")
< 
---
>     # strG0_Initial_Z=("G0 Z" + format(float(CURRENT_Z.getValueAs(UNIT_FORMAT)), strFormat) + "\n")
>     start_Z = CURRENT_Z
658,659c658,665
< #    if DRILL_RETRACT_MODE == "G98" and CURRENT_Z >= RETRACT_Z:
< #        RETRACT_Z = CURRENT_Z
---
>     # wolfgangr 2024-03-24 - try to implement correct G98
>     # see https://github.com/FreeCAD/FreeCAD/issues/13106#issuecomment-2016712284
>     if DRILL_RETRACT_MODE == "G98":
>       final_Z = start_Z
>     elif DRILL_RETRACT_MODE == "G99":
>       final_Z = RETRACT_Z
>     else:
>       final_Z = False
678a685,688
>         strG0_final_Z = (
>             "G0 Z" + format(float(final_Z.getValueAs(UNIT_FORMAT)), strFormat) + "\n"
>         )
> 
718,719c728,731
<             trBuff += linenumber() + strG0_RETRACT_Z
<         else:  # 'G83'
---
> 
>             # return as configured by DRILL_RETRACT_MODE
>             trBuff += linenumber() + strG0_final_Z
>         elif cmd == "G83":  # 'G83'
755,758c767
<                         if DRILL_RETRACT_MODE == "G98" :
<                             trBuff += (linenumber() + strG0_Initial_Z)
<                         else:
<                             trBuff += (linenumber() + strG0_RETRACT_Z)
---
>                         trBuff += (linenumber() + strG0_final_Z)
```

wolfgangr, 2024-03-25:
next step: **rapid moves** in the cycles

- all XY moves 
- all Z moves upwards
- all Z moves downwards above **"Safe Height"**
- Z moves downwards in pecking operations close to previous level

let's collect the speeds:

- Z feedrate from param "F" in canned G 
- XY rapid rate from previous positioning, collected in global LAST_RAPID_XY
- Z rapid rate from initial positioning, collected in global LAST_RAPID_Z

so we need these levels

- "Clearance Height" - from G98/G99 as developed above
- safe_Z - ### to be done ####
- drill_Z - param "Z" in canned G
- retract_Z - param "Z" in canned G

In the current master branch, and in the default settings of the GUI, safe_Z ist set equal to retract_Z. This may be OK for G81, G82, G73. However, for G83, retract levels below stock surface should be possible. If we can't pull variables from the GUI, we may add a SAFE_HEIGHT parameter instead. Keeping this to default "False" disables rapid z move down to safe height.

-----
see also:
https://wiki.freecad.org/CAM_Drilling#Properties
https://wiki.freecad.org/CAM_SetupSheet
CAM Forum: [How can I access Start Depth, Safe Height in a postprocessor?](https://forum.freecad.org/viewtopic.php?t=86391)

wolfgangr, 2024-03-25:
After understanding and rearranging the G83 code, it looks like it does what it's supposed to, now. 

In the simulation I really feel like turning the feed wheel manually. Can we drill steel on a TTC450 with such kind of code :-))) ?

(G98 clear level = 10, safe level = 1, retract level = -1, Q=1, Z = -6)

```
(Drilling)
(Begin Drilling)
G0 Z10.000 F500.000
G90
( G98 )
G0 X10.000 Y-1.000 F2000.000
(G83 X10.000 Y-1.000 Z-6.000 F100.000 Q1.000 R-1.000)
G0 X10.000 Y-1.000 F2000.00  # jog xy at G98 clear height
G0 Z1.000 F500.00            # rapid down to safe_Z
G1 Z0.000 F100.00            # 1st feed
G0 Z0.050 F500.00            # up just a_bit
G1 Z-1.000 F100.00           # 2nd feed
G0 Z-0.950 F500.00           # up a_bit
G1 Z-2.000 F100.00           # 3rd feed
G0 Z-1.000 F500.00           # now to retract_Z
G0 Z-1.950 F500.00           # rapid down till a_bit above
G1 Z-3.000 F100.00           # 4th feed
G0 Z-1.000 F500.00
G0 Z-2.950 F500.00
G1 Z-4.000 F100.00
G0 Z-1.000 F500.00
G0 Z-3.950 F500.00
G1 Z-5.000 F100.00
G0 Z-1.000 F500.00
G0 Z-4.950 F500.00
G1 Z-6.000 F100.00           # last feed
G0 Z10.000 F500.00           # finish cycle at G98 clear height
G0 X24.000 Y10.000 F2000.000
```

We see that above retract_Z, the pecking is the same as required for G73, so all we need to implement G73 as a variant of G83 ist an expansion of a single if-clause:

```
if ((cmd == "G83") and clearance_depth < RETRACT_Z):
```

Just have a look whether we can tidy down the code a bit further and lump G81 and G82 in as well.

wolfgangr, 2024-03-25:
try to add file ...

wolfgangr, 2024-03-25:
tried to attach file here, does not work
attached it in the forum instead
[grbl_post.py](https://forum.freecad.org/download/file.php?id=258735)

wolfgangr, 2024-03-28:
**Tested - works.** No problems keeping this settings in mixed projects of drilling and milling operations. Have it in my default settings now  and will leave a message here when I encounter issues.

Just began to understand how the milling operations take their heights from Operation Op/Height/Depth property values. To optimize overall job performance keep Safe Height an Cleareance Height low and in sync. I've seen that e.g. the pad dressup creates really many ugly slow moves, but that's beyond the issue here at hand.

Therefore, it would be nice to bind my `--safe_value` to the Operations params. No response my [forum post](https://forum.freecad.org/viewtopic.php?t=86391) yet. 

--------------------------

my personal issue state: **"works for me".** 

I'll keep this open for the moment and ask the community if there is any proper next step to contribute.

ShamanTcler, 2025-01-15:
Is this issue for both the grpl_post.py and the refactored_grpl_post.py? I believe that lwoestman and I would prefer to see a migration to the refactored versions.

wolfgangr, 2025-02-25:
sorry, haven't been involved in FreeCAD for quite some time now.

As far as I can remember, I only had a short glance in the "refactored" code.
No clue whether the problem is already solved there.

If there is demand in the community, and "refactured" is the path to future, I'll take a look.

There's another issue that might be handled at that point:

grbl obviusly considers small arcs with huge radii as circles, which may lead mill astraying off bed for models generated from "unclean" sources:
https://forum.freecad.org/viewtopic.php?style=4&p=792166#p792166

J-Dunn, 2025-07-16:
> Is this issue for both the grpl_post.py and the refactored_grpl_post.py? 

I think one of the criteria for refactored*post was that the output was identical.  It was only intended to be a refactoring.

J-Dunn, 2025-07-16:
I did the G83 expansion for GRBL. At that time where was only G83 in FreeCAD so that is the only peck which was expanded. 

There is a bug in that code which expands G81,82 then drops through to G83, so G73 is getting treated as G83 instead of producing nothing since it does not expand it.  GRBL will ignore G73 . 

So it's not a total flop despite FC switching to G73. 

G73 seems to be some kind of non standard extra which is implemented differently on several machines which do recognise it.  

I'm currently trying to find out when this was introduced. 

>  It may be generated by the Path tool in certain circumstances (may be shallow holes and peck drill enabled?)

I was under the impression it had simple replaced G83. Is there someone trying to be too clever and trying to think what the user "needs" to be doing here?

 Can anyone find when G73 was added?

J-Dunn, 2025-07-16:
`--translate_drill --preamble "G17 G90 G0 F3000 "  --rapidspeed --safe_height 2`

IMO, we should not be trying to set these parameters in post-proc.  They should be correctly implemented in the work bench and Gcode should contain all the information needed already. 

Expanding for grbl should not require anything which is not already in the gcode.  Having to enter this twice means there is a deeper problem.

J-Dunn, 2025-07-16:
Ah, I've just found where we already discussed this. 
https://forum.freecad.org/viewtopic.php?t=86391#p750379

I still need to know where this G73 stuff came in.