FreeCAD/FreeCAD #13107: part design: additive loft crash

kinghat, 2024-03-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

additive loft with update view enabled is not catching the otherwise handled error below and segfaults instead:

> AdditiveLoft002: Loft: Sections need to have the same amount of inner wires (except profile and last section, which can be points)

- open file
- double click into loft
- add sketch002 with update view enabled

if you dont have update view enabled and add the sketch and close out the error is handled and displayed for you.

### Full version info

```shell
OS: Fedora Linux 39 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git) AppImage
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * Curves 0.6.29
  * CurvedShapes 1.0.8
```

### Subproject(s) affected?

PartDesign

### Anything else?

```
$ ./FreeCAD_weekly-builds-36487-2024-03-19-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR36487 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3e9a0) [0x7f3bfe45c9a0]
#1  0x7f3bfe8bf08b in QMetaObject::cast(QObject const*) const from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0xb
#2  0x7f3bffa3b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x16
#3  0x7f3c015566e7 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x4e7
#4  0x7f3c0155686f in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x12f
#5  0x7f3c015746d4 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x134
#6  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f3bfe8eb6f2]
#7  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5(+0x43cc76) [0x7f3bffa3cc76]
#8  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7f3bfe8eb964]
#9  0x7f3bfe864ce7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x27
#10  0x7f3bfe86d08e in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x1ee
#11  0x7f3bffa318e8 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x688
#12  0x7f3bffa321b7 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x167
#13  0x7f3bff9c3423 in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x2e3
#14  0x7f3c01582e18 in Gui::TreeWidget::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x28
#15  0x7f3bff7aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x178
#16  0x7f3bff856c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#17  0x7f3bfe8b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x7b
#18  0x7f3bff77039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#19  0x7f3bff778262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#20  0x7f3c01327509 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x99
#21  0x7f3bfe8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x108
#22  0x7f3bff7763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#23  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f3bff7c6e6f]
#24  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f3bff7ca0c9]
#25  0x7f3bff7703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#26  0x7f3c01327509 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x99
#27  0x7f3bfe8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x108
#28  0x7f3bfef353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Gui.so.5+0x84c
#29  0x7f3bfef0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Gui.so.5+0xac
#30  /tmp/.mount_FreeCAJiskxw/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f3bec09fbea]
#31  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f3bfcd10ad7]
#32  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f3bfcd13ac7]
#33  /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f3bfcd140ce]
#34  0x7f3bfe90aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x66
#35  0x7f3bfe8b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x11b
#36  0x7f3bfe8bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0x82
#37  0x7f3c0128ad67 in Gui::Application::runApplication() from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0xa27
#38  /tmp/.mount_FreeCAJiskxw/usr/bin/freecad(+0x7620) [0x560e38245620]
#39  /lib64/libc.so.6(+0x2814a) [0x7f3bfe44614a]
#40  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f3bfe44620b]
#41  /tmp/.mount_FreeCAJiskxw/usr/bin/freecad(+0x7981) [0x560e38245981]
```

[additive_loft_crash_test.zip](https://github.com/FreeCAD/FreeCAD/files/14730102/additive_loft_crash_test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-03-25:
I don't get segmentation fault, only the error is shown. _Update view_ disabled/enabled does not cause crash on my system.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36520 (Git)
Build type: Debug
Branch: main
Hash: 1cfad4142275f7cd25bcc32bea95e88f510b8246
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

kinghat, 2024-04-04:
tested again with the new weekly build and still get the segfault:

```
$ ./FreeCAD_weekly-builds-36731-2024-04-02-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR36731 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3e9a0) [0x7ffaf205c9a0]
#1  0x7ffaf515662e in Gui::DocumentObjectItem::setCheckState(bool) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x9e
#2  0x7ffaf51593d2 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x4f2
#3  0x7ffaf515954f in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x12f
#4  0x7ffaf5180944 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x134
#5  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7ffaf24eb6f2]
#6  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5(+0x43cc76) [0x7ffaf363cc76]
#7  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7ffaf24eb964]
#8  0x7ffaf2464ce7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5+0x27
#9  0x7ffaf246d08e in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5+0x1ee
#10  0x7ffaf36318e8 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x688
#11  0x7ffaf36321b7 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x167
#12  0x7ffaf35c3423 in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x2e3
#13  0x7ffaf5185fa8 in Gui::TreeWidget::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x28
#14  0x7ffaf33aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x178
#15  0x7ffaf3456c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#16  0x7ffaf24b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5+0x7b
#17  0x7ffaf337039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#18  0x7ffaf3378262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#19  0x7ffaf4f29339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x99
#20  0x7ffaf24b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5+0x108
#21  0x7ffaf33763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#22  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7ffaf33c6e6f]
#23  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7ffaf33ca0c9]
#24  0x7ffaf33703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#25  0x7ffaf4f29339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x99
#26  0x7ffaf24b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5+0x108
#27  0x7ffaf2b353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Gui.so.5+0x84c
#28  0x7ffaf2b0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Gui.so.5+0xac
#29  /tmp/.mount_FreeCAp6MRE3/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7ffadfb94bea]
#30  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7ffaf0b10ad7]
#31  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7ffaf0b13ac7]
#32  /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7ffaf0b140ce]
#33  0x7ffaf250aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libQt5Core.so.5+0x66
[... cut ...]
#40  /tmp/.mount_FreeCAp6MRE3/usr/bin/freecad(+0x7981) [0x559e790eb981]
```

> OS: Fedora Linux 39 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.36731 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
> Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)
> Installed mods: 
>   * OpenDark 2023.12.17
>   * Curves 0.6.29
>   * CurvedShapes 1.0.8

i will test in a fresh vm later and report back.

kinghat, 2024-04-04:
tested in fresh gnome vm:

```
$ ./FreeCAD_weekly-builds-36731-2024-04-02-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR36731 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3e9a0) [0x7f76dfe5c9a0]
#1  0x7f76e13be2d6 in QAbstractItemView::model() const from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x6
#2  0x7f76e143afa1 in QTreeWidgetItem::treeModel(QTreeWidget*) const from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x11
#3  0x7f76e143b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x16
#4  0x7f76e2f593c7 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libFreeCADGui.so+0x4e7
#5  0x7f76e2f5954f in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libFreeCADGui.so+0x12f
#6  0x7f76e2f80944 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libFreeCADGui.so+0x134
#7  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f76e02eb6f2]
#8  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5(+0x43cc76) [0x7f76e143cc76]
#9  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7f76e02eb964]
#10  0x7f76e0264ce7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5+0x27
#11  0x7f76e026d08e in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5+0x1ee
#12  0x7f76e14318e8 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x688
#13  0x7f76e14321b7 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x167
#14  0x7f76e13c3423 in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x2e3
#15  0x7f76e2f85fa8 in Gui::TreeWidget::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libFreeCADGui.so+0x28
#16  0x7f76e11aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x178
#17  0x7f76e1256c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#18  0x7f76e02b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5+0x7b
#19  0x7f76e117039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#20  0x7f76e1178262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#21  0x7f76e2d29339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libFreeCADGui.so+0x99
#22  0x7f76e02b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5+0x108
#23  0x7f76e11763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#24  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f76e11c6e6f]
#25  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f76e11ca0c9]
#26  0x7f76e11703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#27  0x7f76e2d29339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libFreeCADGui.so+0x99
#28  0x7f76e02b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Core.so.5+0x108
#29  0x7f76e09353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Gui.so.5+0x84c
#30  0x7f76e090b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Gui.so.5+0xac
#31  /tmp/.mount_FreeCAR9Wcdj/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f76cda15bea]
#32  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f76de910ad7]
#33  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f76de913ac7]
#34  /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f76de9140ce]
[... cut ...]
#40  /lib64/libc.so.6(+0x2814a) [0x7f76dfe4614a]
#41  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f76dfe4620b]
#42  /tmp/.mount_FreeCAR9Wcdj/usr/bin/freecad(+0x7981) [0x5579fd791981]
```

> OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.36731 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
> Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)

xtemp09, 2024-04-05:
@kinghat, could you record a video to show the exact steps to reproduce the crash? You have provided the three backtrace logs, however, if you look at them, they slightly differ in the end.

```cpp
#1  0x7f3bfe8bf08b in QMetaObject::cast(QObject const*) const from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0xb
#2  0x7f3bffa3b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x16
#3  0x7f3c015566e7 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x4e7
```

```cpp
#1  0x7ffaf515662e in Gui::DocumentObjectItem::setCheckState(bool) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x9e
#2  0x7ffaf51593d2 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x4f2
#3  0x7ffaf515954f in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x12f
```

```cpp
#1  0x7f76e13be2d6 in QAbstractItemView::model() const from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x6
#2  0x7f76e143afa1 in QTreeWidgetItem::treeModel(QTreeWidget*) const from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x11
#3  0x7f76e143b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x16
```

It looks like each crash caused by different actions.

kinghat, 2024-04-05:
> @kinghat, could you record a video to show the exact steps to reproduce the crash? You have provided the three backtrace logs, however, if you look at them, they slightly differ in the end.
> 
> ```c++
> #1  0x7f3bfe8bf08b in QMetaObject::cast(QObject const*) const from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Core.so.5+0xb
> #2  0x7f3bffa3b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libQt5Widgets.so.5+0x16
> #3  0x7f3c015566e7 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAJiskxw/usr/bin/../lib/libFreeCADGui.so+0x4e7
> ```
> 
> ```c++
> #1  0x7ffaf515662e in Gui::DocumentObjectItem::setCheckState(bool) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x9e
> #2  0x7ffaf51593d2 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x4f2
> #3  0x7ffaf515954f in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAp6MRE3/usr/bin/../lib/libFreeCADGui.so+0x12f
> ```
> 
> ```c++
> #1  0x7f76e13be2d6 in QAbstractItemView::model() const from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x6
> #2  0x7f76e143afa1 in QTreeWidgetItem::treeModel(QTreeWidget*) const from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x11
> #3  0x7f76e143b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAR9Wcdj/usr/bin/../lib/libQt5Widgets.so.5+0x16
> ```
> 
> It looks like each crash caused by different actions.

[Screencast from 2024-04-05 12-52-28.webm](https://github.com/FreeCAD/FreeCAD/assets/16073360/cbb2c863-2cc6-4021-b9b2-cb106b540dd1)

```
$ ./FreeCAD_weekly-builds-36731-2024-04-02-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR36731 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3e9a0) [0x7f2739a5c9a0]
#1  0x7f273a7be2d6 in QAbstractItemView::model() const from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x6
#2  0x7f273a83afa1 in QTreeWidgetItem::treeModel(QTreeWidget*) const from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x11
#3  0x7f273a83b436 in QTreeWidgetItem::setSelected(bool) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x16
#4  0x7f273c3593c7 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libFreeCADGui.so+0x4e7
#5  0x7f273c35954f in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libFreeCADGui.so+0x12f
#6  0x7f273c380944 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libFreeCADGui.so+0x134
#7  /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f27396eb6f2]
#8  /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5(+0x43cc76) [0x7f273a83cc76]
#9  /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7f27396eb964]
#10  0x7f2739664ce7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Core.so.5+0x27
#11  0x7f273966d08e in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Core.so.5+0x1ee
#12  0x7f273a8318e8 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x688
#13  0x7f273a8321b7 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x167
#14  0x7f273a7c3423 in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x2e3
#15  0x7f273c385fa8 in Gui::TreeWidget::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libFreeCADGui.so+0x28
#16  0x7f273a5aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x178
#17  0x7f273a656c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#18  0x7f27396b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Core.so.5+0x7b
#19  0x7f273a57039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#20  0x7f273a578262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#21  0x7f273c129339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAg6bgta/usr/bin/../lib/libFreeCADGui.so+0x99
[... cut ...]
#39  /tmp/.mount_FreeCAg6bgta/usr/bin/freecad(+0x7620) [0x5624bd68d620]
#40  /lib64/libc.so.6(+0x2814a) [0x7f2739a4614a]
#41  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f2739a4620b]
#42  /tmp/.mount_FreeCAg6bgta/usr/bin/freecad(+0x7981) [0x5624bd68d981]
```

> OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git) AppImage
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

xtemp09, 2024-04-06:
This is a _Qt_ bug. The crash is most likely caused by
https://bugreports.qt.io/browse/QTBUG-94546

This comment says it is to be fixed in _Qt 7_:
https://github.com/qt/qtbase/blob/772fd609c6eff09bdaa44cb2ac5f913e58788fa3/src/widgets/itemviews/qtreewidget.h#L184-L185

These bugs are closely related:
https://bugreports.qt.io/browse/QTBUG-93829
https://bugreports.qt.io/browse/QTBUG-88003
https://bugreports.qt.io/browse/QTBUG-55402

It looks like one can't do anything, lest completely rewrite `Tree.cpp` and `Tree.h`. 😢

xtemp09, 2024-04-07:
If Qt guys ever come across this topic, here is the backtrace.

<details><summary>The backtrace</summary>

```cpp
Process 1704 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x3b)
    frame #0: 0x00007ffff3242909 libQt5Widgets.so.5`QTreeModel::index(QTreeWidgetItem const*, int) const [inlined] QListData::size(this=0x000000000000003b) const at qlist.h:115:51
   112      void remove(int i);
   113      void remove(int i, int n);
   114      void move(int from, int to);
-> 115      inline int size() const noexcept { return int(d->end - d->begin); }   // q6sizetype
   116      inline bool isEmpty() const noexcept { return d->end  == d->begin; }
   117      inline void **at(int i) const noexcept { return d->array + d->begin + i; }
   118      inline void **begin() const noexcept { return d->array + d->begin; }
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x3b)
  * frame #0: 0x00007ffff3242909 libQt5Widgets.so.5`QTreeModel::index(QTreeWidgetItem const*, int) const [inlined] QListData::size(this=0x000000000000003b) const at qlist.h:115:51
    frame #1: 0x00007ffff3242909 libQt5Widgets.so.5`QTreeModel::index(QTreeWidgetItem const*, int) const [inlined] QList<QTreeWidgetItem*>::count(this=0x000000000000003b) const at qlist.h:359:45
    frame #2: 0x00007ffff3242909 libQt5Widgets.so.5`QTreeModel::index(this=0x0000555555c63640, item=0x0000555559214070, column=0) const at qtreewidget.cpp:243:31
    frame #3: 0x00007ffff3242e08 libQt5Widgets.so.5`QTreeWidgetItem::setSelected(this=0x0000555559214070, select=<unavailable>) at qtreewidget.cpp:1022:43
    frame #4: 0x00007ffff6a813df libFreeCADGui.so`Gui::DocumentItem::updateItemSelection(this=0x0000555557ff9b60, item=0x0000555559214070) at Tree.cpp:4668:30
    frame #5: 0x00007ffff6a80c62 libFreeCADGui.so`Gui::DocumentItem::clearSelection(this=0x0000555557ff9b60, exclude=0x0000555559214070) at Tree.cpp:4552:28
    frame #6: 0x00007ffff6a774ba libFreeCADGui.so`Gui::TreeWidget::onItemSelectionChanged(this=0x0000555555c924e0) at Tree.cpp:3318:37
    frame #7: 0x00007ffff6ab0c5a libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TreeWidget::*)()>::call(f=(libFreeCADGui.so`Gui::TreeWidget::onItemSelectionChanged() at Tree.cpp:3258:1), o=0x0000555555c924e0, arg=0x00007fffffffb630) at qobjectdefs_impl.h:152:20
    frame #8: 0x00007ffff6aa8495 libFreeCADGui.so`void QtPrivate::FunctionPointer<void (Gui::TreeWidget::*)()>::call<QtPrivate::List<>, void>(f=(libFreeCADGui.so`Gui::TreeWidget::onItemSelectionChanged() at Tree.cpp:3258:1), o=0x0000555555c924e0, arg=0x00007fffffffb630) at qobjectdefs_impl.h:185:95
    frame #9: 0x00007ffff6a9ec15 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::TreeWidget::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x0000555555c99f20, r=0x0000555555c924e0, a=0x00007fffffffb630, ret=0x0000000000000000) at qobjectdefs_impl.h:418:49
    frame #10: 0x00007ffff22e33db libQt5Core.so.5`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(a=0x00007fffffffb630, r=0x0000555555c924e0, this=0x0000555555c99f20) at qobjectdefs_impl.h:398:57
    frame #11: 0x00007ffff22e33bf libQt5Core.so.5`void doActivate<false>(sender=0x0000555555c924e0, signal_index=25, argv=0x00007fffffffb630) at qobject.cpp:3925:30
    frame #12: 0x00007ffff22da3ab libQt5Core.so.5`QMetaObject::activate(sender=0x0000555555c924e0, m=0x000000000073f640, local_signal_index=9, argv=0x0000000000000000) at qobject.cpp:3985:26
    frame #13: 0x00007ffff32409ea libQt5Widgets.so.5`QTreeWidget::itemSelectionChanged(this=0x0000555555c924e0) at moc_qtreewidget.cpp:451:26
    frame #14: 0x00007ffff3242843 libQt5Widgets.so.5`QTreeWidgetPrivate::_q_selectionChanged(this=<unavailable>, selected=<unavailable>, deselected=0x00007fffffffb918) at qtreewidget.cpp:2477:33
    frame #15: 0x00007ffff324aed2 libQt5Widgets.so.5`QTreeWidget::qt_static_metacall(_o=<unavailable>, _c=<unavailable>, _id=<unavailable>, _a=<unavailable>) at moc_qtreewidget.cpp:217:51
    frame #16: 0x00007ffff22e357c libQt5Core.so.5`void doActivate<false>(sender=0x0000555555c93aa0, signal_index=3, argv=0x00007fffffffb7e0) at qobject.cpp:3937:33
    frame #17: 0x00007ffff22da3ab libQt5Core.so.5`QMetaObject::activate(sender=0x0000555555c93aa0, m=0x00000000005a3aa0, local_signal_index=0, argv=0x00007fffffffb7e0) at qobject.cpp:3985:26
    frame #18: 0x00007ffff225564a libQt5Core.so.5`QItemSelectionModel::selectionChanged(this=0x0000555555c93aa0, _t1=0x00007fffffffb928, _t2=0x00007fffffffb918) at moc_qitemselectionmodel.cpp:475:26
    frame #19: 0x00007ffff225adb4 libQt5Core.so.5`QItemSelectionModel::emitSelectionChanged(this=0x0000555555c93aa0, newSelection=0x00007fffffffb928, oldSelection=0x00007fffffffb918) at qitemselectionmodel.cpp:1886:30
    frame #20: 0x00007ffff225cf7a libQt5Core.so.5`QItemSelectionModel::select(this=0x0000555555c93aa0, selection=<unavailable>, command=(i = 35)) at qitemselectionmodel.cpp:1340:25
    frame #21: 0x00007ffff3235e0c libQt5Widgets.so.5`QTreeViewPrivate::select(this=0x0000555555cabc20, topIndex=<unavailable>, bottomIndex=<unavailable>, command=<unavailable>) at qtreeview.cpp:3927:32
    frame #22: 0x00007ffff3236075 libQt5Widgets.so.5`QTreeView::setSelection(this=0x0000555555c924e0, rect=<unavailable>, command=(i = 35)) at qtreeview.cpp:2325:14
[... cut ...]
    frame #67: 0x000055555557eca5 FreeCAD`_start + 37
(lldb) 

```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Debug
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

The crash takes place at this line:

https://github.com/qt/qtbase/blob/66dd00dcfd8a715ff6bc736e6beb5f2d3b2c69cc/src/widgets/itemviews/qtreewidget.cpp#L206

wwmayer, 2024-04-27:
I am not sure if this is really a Qt bug. The stack trace is:
1. TreeWidget::onItemSelectionChanged()
2. DocumentItem::clearSelection
3. DocumentItem::updateItemSelection
At line 4665 item is still a valid pointer. Then at line 4666 Gui::Selection().addSelection is called that notifies its observers and e.g. calls TaskLoftParameters::onSelectionChanged. This triggers a recompute of the shape and re-creates all tree items and thus deletes the item above. Then at line 4667 (`item->selected = 0;`) item is a dangling pointer and causes the crash.

wwmayer, 2024-04-27:
Part of the problem is the function ViewProviderSuppressibleExtension::extensionUpdateData() because it emits the signal signalHighlightObject that triggers the recreation of the tree items. The second part of the problem is that the slot function forces an immediate update of the tree -- needs to be checked if delaying the update is the better way.

wwmayer, 2024-04-27:
So, the two main problems are the methods:
1. DocumentItem::slotHighlightObject: This calls TreeWidget::_updateStatus() to perform an immediate update that may destroy and recreate all items. However, it's not clear why the update is needed here at all.  The call of _updateStatus() has been added with c18bf1182
2.  DocumentItem::updateItemSelection: When calling Gui::Selection().addSelection() a lot of things can happen that can be hardly controlled and this can lead to unexpected side-effects like the destruction of the used item.

FlachyJoe, 2024-04-27:
@wwmayer  #13682 fix ViewProviderSuppressibleExtension::extensionUpdateData() so the bug is not triggered anymore but the underlying problems remain.

wwmayer, 2024-04-27:
Emitting the signal ViewProviderSuppressibleExtension::extensionUpdateData() just reveals the bug that exists in the tree view code. Replacing the signal with something else cures the symptoms but it doesn't fix the bug.

And who knows which other crashes happen for the exact same reason.

wwmayer, 2024-04-27:
Here my PR https://github.com/FreeCAD/FreeCAD/pull/13686

commit 219a2e01 (2024-05-13): Fix #13107 (#13682)

Introduce a signal slot for tree item highlight change.
closed 2024-05-13 by chennes