FreeCAD/FreeCAD #13108: name 'Gui' is not defined

josePereiro, 2024-03-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, the error occur when trying to add a distance constraint. The dialog window appears, but after introducing the distance and hitting ok the constraint fail to be added and the bellow error is printed.

```
23:44:49  Traceback (most recent call last):
  File "<string>", line 12, in <module>
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Show/mTempoVis.py", line 443, in sketchClipPlane
    editDoc = Gui.editDocument()
<class 'NameError'>: name 'Gui' is not defined
23:44:49  setEdit: visibility automation failed with an error: 
23:45:54  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 13.80 mm for the constraint with index 13 is invalid
23:45:54  App.getDocument('CSC_Reactor').getObject('Sketch').setDatum(13,App.Units.Quantity('13.800000 mm'))
```

The error is reproducible after I close and reopen FreeCAD. 

Just in case, [here](https://github.com/FreeCAD/FreeCAD/files/14730191/Archive.zip) the project's file. I'm trying to add the constraint between the circle and the top left corner of the rectangle.

### Full version info

```shell
[code]
OS: macOS 13.0.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-03-23:
`Gui` is defined in the file:
https://github.com/FreeCAD/FreeCAD/blob/edbab63c916d74eb31df922a27a5878a02efda68/src/Mod/Show/mTempoVis.py#L31-L32

But we have seen this code fail on macOS before:
https://forum.freecad.org/viewtopic.php?f=3&t=60251

luzpaz, 2024-08-24:
@josePereiro is this still an issue on the dev build ? Can you test ?

luzpaz, 2024-08-26:
Closing as no reproducing on dev version shown

closed 2024-08-26 by luzpaz