FreeCAD/FreeCAD #13118: Support HiDPI Screens on Windows

maxwxyz, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While there is some basic support for HiDPI screens on Windows in 0.22dev (0.21.2 looks awfully blurred) it is still a problem, when I dock my HiDPI screen to a non-HiDPI external monitor in Windows 11. FreeCAD won't look the same on both screens and it looks wrongly scaled on each of them. So currently it is hard to work in FreeCAD on a multiple monitor setup when using different resolutions on each screen, although you only try to run it on one of them (not using both the same time).
Overlays and icons are rendered huge, also in all panels (task, tree,...). Cursor/pointer icons are either blurry or pixelated, as they don't account for the screen resolution (only works in Sketcher WB). All dialogs with buttons render hugely and text is not legible.

There is no problem when I only use the HiDPI screen on the Laptop or plug in a normal DPI Laptop to the same normal DPI external screen, so I think the combination of different scaled screens messes some things up.

Default view on the HiDPI screen while plugged in another normal DPI screen:
[image]

Preferences dialog is larger than the screen and will leak over to the normal DPI screen where it is huge (@kadet1090  FYI):
[image]

@chennes maybe the recent addon manager could be fixed in the same go:
[image]

When opening FreeCAD on the external normal DPI monitor, same result but some items are scaled up too large:
[image]

@obelisk79 @MisterMakerNL  is there a workaround for the themes, they are currently also hard to use.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.3
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.15
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.6
  * sheetmetal.backup1710961794.9712274 0.4.3 (Disabled)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-06:
For an added comparison: It was no issue in 0.21.2 stable.
Compare both, left is 21.2, right is 0.22
[image]

maxwxyz, 2024-04-06:
That is my full screen size on the HiDPI screen with the preferences dialog:
[image]

And it continues huge on the normal DPI screen:
[image]

No issue in 0.21.2

xtemp09, 2024-04-07:
> `Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2`

@maxwxyz, could you test _FreeCAD_ with _Qt 6_? _Qt 6_ is slightly different from _Qt 5_ in Per-Monitor DPI Awareness.[^1]

[^1]: https://doc.qt.io/qt-6/highdpi.html#configuring-windows

maxwxyz, 2024-04-07:
@xtemp09 how can I do that?

xtemp09, 2024-04-07:
I've never compiled _FreeCAD_ on Windows, so I don't have any idea. :'(

maxwxyz, 2024-05-22:
I guess this is related to this issue here:
The BIM Welcome screen is not ideally sized and text rendered too big
[image]
@yorikvanhavre FYI

maxwxyz, 2024-06-27:
@kadet1090 is there a chance this could be solved for 1.0?

maxwxyz, 2024-07-25:
@chennes would you consider this a blocker for 1.0 or is this trivial and solved with a Qt6 build?

maxwxyz, 2024-07-30:
@chennes any progress on this one? As it seems that Windows will not be released with Qt6 there is no fix in 1.0 to work correctly on HiDPI screens on Windows?

chennes, 2024-07-31:
Qt5's HiDPI support is basically terrible, and anything we do to make it better on one system is likely to make it worse on another (e.g. how it behaves on multiple monitors, etc.). It's going to be a hard sell convincing a developer to spend their time on this problem since it's effectively solved by switching to Qt6. What we could consider doing is a simultaneous release of an "experimental" Qt6 build of 1.0 on Windows. It's likely to cause other issues (though I've been using it without too much trouble for the last month).

cbenhagen, 2024-07-31:
HiDPI currently causes troubles on macOS as well. Maybe such an experimental build with QT6 would be a great idea on all platforms.

maxwxyz, 2024-07-31:
There are still a lot of issues with Qt6 on Windows as it looks completely different on Windows 11 than Windows 10.
The thread starts about here: https://discord.com/channels/870877411049357352/870883914191888444/1267923058614735089
And maintaining a version to fit all (Qt5, Qt6, Win10, Win11,... + other OS) seems to be a huge effort.

MisterMakerNL, 2024-07-31:
It's just setting Qstyle too fusion.
If someone wants to design a theme specifically for windows 11, then have fun with that. :)

luzpaz, 2024-08-30:
> It's just setting Qstyle too fusion.

* @MisterMakerNL is this what you're referring to? #15868

MBach, 2025-01-18:
[image]

xtemp09, 2025-01-19:
@MBach, could you provide screenshots with different `QT_SCALE_FACTOR`, say from _1.25_ to _2_? Does you system use scaling factors already?

MBach, 2025-01-19:
> [@MBach](https://github.com/MBach), could you provide screenshots with different `QT_SCALE_FACTOR`, say from _1.25_ to _2_? Does you system use scaling factors already?

How do I set this variable? I've just downloaded and installed latest binaries from the website? I've tried creating a shortcut, then set QT_SCALE_FACTOR=2 but I'm not sure this is the correct way to proceed.

xtemp09, 2025-01-20:
@MBach, ok, now can you report your OS, screen resolution and if your OS uses scaling? It would be good if you provide the _About_ info.

@MBach, if your system does not use scaling, I would set `QT_SCALE_FACTOR` to _1.25_ or _1.4_.

MBach, 2025-01-20:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

luzpaz, 2025-04-20:
Is this still an issue on Windows w/ Qt6 ?

closed 2025-05-21 by maxwxyz

adrianinsaval, 2025-05-22:
> No issue with Qt6, will 1.1 be released with Qt6 [@adrianinsaval](https://github.com/adrianinsaval) ?

late response but yes