FreeCAD/FreeCAD #13119: Request to clearly separate exiting sketch tools and exiting Sketcher

katepavlu, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**The problem**
When using Sketcher, tools such as Create Line or Constrain Vertically need to be exited before one can manipulate sketch entities by dragging. Exiting a tool is performed by pressing ESC.

Unfortunately, ESC is also the keyboard shortcut for exiting the Sketch altogether. This means that long pressing the key, pressing it twice, or pressing it when no tool is actually selected, closes the sketch. The user has to reopen it, disrupting workflow. 

**When does this happen**
- When the user is confused on whether they have a tool selected or not. This happens to me sometimes, since the _selected tool indicator is small and not very high contrast_ against the blue gradient background. I have a small visual impairment, which might be making this problem slightly worse.
- With a faulty keyboard
- When an older user with slower reaction speed accidentally holds the key too long. I have taught the program to an older coworker, this was a problem sometimes.
- When the user is used to other CAD programs not doing this and has the habit of pressing ESC before manipulating sketch entities "just in case"

**What do other CADs do**
Professional tools avoid this problem by having different UI for exiting tools and exiting the Sketch, requiring the user to click some exit button. This is the case for SOLIDWORKS, Fusion, Onshape and, if my memory serves me correctly, also Creo Parametric.

**Proposed solutions**
- Remove the keyboard shortcut for exiting Sketcher, since we already have a button to do it. This would bring UI/UX closer to industry standard.
- Remap the shortcut to something else, eg. Shift+ESC
- Increase size and or contrast of the selected tool indicators. This would fix the problem in some but not all cases.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
Also tested on Windows (FreeCAD 0.20, 0.21).
Also tested with Ondsel (2024.1).
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-24:
It could also be great if it would be displayed, if a current sketch tool is still active. Sometimes, on my system, the mouse pointer switches to the default one, although the current tool is still active, mostly with the Dimension tool (@PaddleStroke).
Then I have to right click before using other tools. Maybe an active tool could be shown in the task panel or with a different color / frame in the 3D view.

Roy-043, 2024-03-24:
> Unfortunately, ESC is also the keyboard shortcut for exiting the Sketch altogether.

There is a preference to prevent that. And using the right mouse button instead is also possible.

katepavlu, 2024-03-25:
> > Unfortunately, ESC is also the keyboard shortcut for exiting the Sketch altogether.
> 
> There is a preference to prevent that. And using the right mouse button instead is also possible.

Thank you for the quick reply! 

I unfortunately cannot find the preference anywhere.
I don't know if long time users rely on this behavior, but if most don't it might be a good idea to change the default, since as far as I know no other CAD tool has the same shortcut for those action 

Right mouse is a good workaround.

Roy-043, 2024-03-25:
> I unfortunately cannot find the preference anywhere.

https://wiki.freecad.org/Sketcher_Preferences#General
Uncheck "Esc can leave sketch edit mode".

katepavlu, 2024-03-25:
> > I unfortunately cannot find the preference anywhere.
> 
> https://wiki.freecad.org/Sketcher_Preferences#General Uncheck "Esc can leave sketch edit mode".

Ah, didn't realize you have to manually load the Sketcher to touch it's settings.
Thank you

benj5378, 2024-10-04:
> https://wiki.freecad.org/Sketcher_Preferences#General Uncheck "Esc can leave sketch edit mode".

Seems like there already exists a solution for this. @maxwxyz, can you close this?

closed 2024-10-05 by luzpaz