FreeCAD/FreeCAD #13121: partdesign: "Cannot subtract primitive feature without a Bbase feature"

annevanleyden, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whenever I change a value in the spreadsheet, the resulting recalculation results in the error "Cylinder069: Cannot subtract primitive feature without base feature", even though the Body was valid when opening the file and no changes were made to the Body.

To reproduce: open the spreadsheet and edit any cell.

The freecad file 
[partdesign-error.zip](https://github.com/FreeCAD/FreeCAD/files/14735442/partdesign-error.zip)

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-wayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36438 (Git) AppImage
Build type: Release
Branch: main
Hash: 0def330a9e2119a14bb349cc406c2aa365421ab4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.4
  * Assembly3 0.11.3
  * Assembly4 0.12.3
  * ThreadProfile 1.83.0
  * fcgear 1.0.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

This bug seems to have been introduced between weekly builds 35456 (not affected) and 35822 (affected).

The original FreeCAD file was a lot bigger, I removed all non-essential stuff to keep the minimum required to reproduce the bug.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-24:
@annevanleyden there seems to be an issue how the body is constructed. First of all, all subtractive and additive cylinders you have used have no attachment. This could be a source of failure. Also when I delete everything incl. the mentioned subtractive cylinder and only leave the first helix and the revolution it also fails. Even just opening your file and hitting recompute fails. The sketch from your revolution is also not properly attached. Not sure if you can try redo your model, as I cannot see where this issue should be.

annevanleyden, 2024-03-24:
@maxwxyz 

Versions before 35822 (or thereabouts) do not exhibit this problem. My Body is fine, without errors or warnings.

How should I, as a user, know about these intricate internals of FreeCAD? If you say my model is corrupt, how and why was I able to create it in the first place?

maxwxyz, 2024-03-24:
can you open the body in the stable version 0.21.2 without problems? I have weird glitches with the subtractive cylinder creating non-solids which could be an OCC problem. https://github.com/FreeCAD/FreeCAD/issues/11995

annevanleyden, 2024-03-24:
"can you open the body in the stable version 0.21.2 without problems?" -> I didn't try that. But did it now. 

I can successfully open the file, edit the Body and make changes. No errors, no warnings.

Syres916, 2024-03-24:
I would be looking at https://github.com/FreeCAD/FreeCAD/commit/a8ae56e06ab0c45205f1f185523c23fe99d5ce44 as a logical first step

Syres916, 2024-03-25:
@annevanleyden the root cause is approx 400 commits before 35822 so we're talking around Xmas time 2023 between:

Good:
``` 13th December 2023
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35370 (Git) AppImage
Build type: Release
Branch: main
Hash: 1066bb54ceec8f4f8376b5149253acda5677651d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.8
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.5
  * Design456 0.0.1
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * AirPlaneDesign 0.4.0
  * fasteners 0.5.10
  * sheetmetal 0.4.1
```

Bad:

``` 15th Jan 2024
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35554 (Git) AppImage
Build type: Release
Branch: main
Hash: 8e2ab9f76888ac4da971f9d5069cec7bcc2b572f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.8
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.5
  * Design456 0.0.1
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * AirPlaneDesign 0.4.0
  * fasteners 0.5.10
  * sheetmetal 0.4.1
```

Syres916, 2024-03-25:
@bgbsww git bisect is showing [bf3ccbc5b69f5b3796056a00c68e92f6e2eb6f5d](https://github.com/FreeCAD/FreeCAD/commit/bf3ccbc5b69f5b3796056a00c68e92f6e2eb6f5d) as responsible for this regression, could you please have a look?

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35531 (Git)
Build type: Release
Branch: (no branch, bisect started on main)
Hash: bf3ccbc5b69f5b3796056a00c68e92f6e2eb6f5d
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.8
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.5
  * Design456 0.0.1
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * AirPlaneDesign 0.4.0
  * fasteners 0.5.10
  * sheetmetal 0.4.1
```

bgbsww, 2024-03-25:
It appears that #12977 replaces the code with the bug, works with the test file, and was coincidentally merged this morning.

maxwxyz, 2024-03-25:
so is it fixed with the new code in #12977 @FlachyJoe

Syres916, 2024-03-27:
> works with the test file

@bgbsww could you please advise which cell you changed & what you changed it to and could you please quote you full info. I have been making the same change to cell _B14_ to _11.8_ in all my tests from the December 2023 build through to current main.
The FreeCAD project still supports OCC 7.6.3 I assume as that's what I'm using and I'm guessing you're not? In order to clarify that another commit hadn't had an unintended consequence, I've just spent another two and half hours bisecting to only find that c7d85ff9d93ad0f1c70fe745ca5838517218642e is corrupting the model and outputting `Transformed: Result has multiple solids. Only keeping the first.` (twice) into the Report View.

@maxwxyz if the FreeCAD project is still supporting OCC 7.6.3 then this is not fixed, it's just a case of wacamole and the result is a new error message (duplicated) and a corrupt model.

I know everyone is doing their best but until the Unit Tests are made **_more_** rigorous (this is a good thing) then the more regressions will get through. IMHO, adding two minutes to the time to test each PR/commit is well worth it versus the hours checking back.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36637 (Git)
Build type: Release
Branch: main
Hash: ef16ecafc25f8505ed6e322d5c2be1c1b4a82f4a
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.8
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.5
  * Design456 0.0.1
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * AirPlaneDesign 0.4.0
  * fasteners 0.5.10
  * sheetmetal 0.4.1
```

bgbsww, 2024-03-27:
First time I ran this on a local build.  I just ran it again on a daily.  Sorry you are having troubles with this; when you flagged it I did drop everything and spent time researching this, only to discover the code had just been replaced.  Helix is unfortunately particularly brutal - no change to it seems to have survived after enough examples appear.  Certainly figuring out more comprehensive testing is warranted, but difficult without examples to work from.

I do see the  `Transformed: Result has multiple solids. Only keeping the first`.  error in the daily, and I don't recall seeing it the first time I was exploring this.  Not sure how we intend to move forward on this.

To your question:  B5 to 61

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36617 (Git)
Build type: Release
Branch: main
Hash: 4845d391ed2702b0b51a9da905066485eb5ef7bc
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * fcgear 1.0.0
  * Plot 2022.4.17
  * DynamicData 2.54.0
  * BIM 2021.12.0
  * parts_library

```