FreeCAD/FreeCAD #13123: Box selection shouldn't require separate tools

FEA-eng, 2024-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, there are two tools to handle box selection in 3D: [Std BoxSelection](https://wiki.freecad.org/Std_BoxSelection) and [Std BoxElementSelection](https://wiki.freecad.org/Std_BoxElementSelection). IMO instead of them, there should be just a built-in user-friendly functionality that allows you to hold LMB (or another button in the case of some specific mouse control schemes) and drag to select. Just like in the Sketcher. Or in other CAD software.

@PaddleStroke Perhaps you'll find this interesting as it fits your recent user-friendliness enhancements.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-24:
Indeed it's a bit weird that box selection is not on click and drag directly.

PaddleStroke, 2024-11-29:
@FreeCAD/design-working-group thoughts about this?

wwmayer, 2024-11-30:
> `Std_BoxSelection` and `Std_BoxElementSelection`

This should be merged into a single command as 99% of the code is identical.

>  Just like in the Sketcher.

Sketcher is very special in this way because it has its own selection logic and bypasses the general selection mechanism. Also, when a sketch is in edit mode there is one explicit view object that handles the selection while with the general box selection there is no such instance.

> there should be just a built-in user-friendly functionality that allows you to hold LMB

What's wrong with having to activate a command to enter box selection mode?

> Or in other CAD software.

I don't know which commercial CAD programs you refer to but according to its documentation the user has to explicitly activate box selection e.g. for CATIA, too. (http://catiadoc.free.fr/online/basug_C2/basugbt0602.htm)

PaddleStroke, 2024-12-01:
> What's wrong with having to activate a command to enter box selection mode?

- It's hidden in a menu. Before FEA-eng opened this issue I wasn't even aware there was a command. So I would venture a guess that this is little known/used.
- Consistency in the app: box selection works by click and drag in sketcher, but in core it needs a command. This is inconsistent.
- Because it is not necessary. Why should it be through a command ? Click and drag is not defined (well it is actually defined in assembly, but we just need to give the priority to handle it to the VP that has edit).
- Click and drag is a behavior that is well known for box selection accross many different softwares.

So overall I don't really see why it should use a command when it could use click and drag?

wwmayer, 2024-12-01:
> It's hidden in a menu.

This can be easily fixed by making the command more prominent in a toolbar or spending its own menu where we can add further options.

> but we just need to give the priority to handle it to the VP that has edit)

But that's exactly the point. When using the box selection usually there is no object in edit mode. And when you talk about consistency it would be very weird if box selection was available if an object is in edit mode and otherwise not.

> So overall I don't really see why it should use a command when it could use click and drag?

Many interactive things with the viewer are done with callback functions that may implement their own event handling and this can easily collide with the box selection that is permanently active.

E.g. look at the measurement objects. When saving the result of a measurement then it is stored in a label shown in the viewer.  This label can be dragged by a mouse click and move -- exactly the same events used for the box selection.

PaddleStroke, 2024-12-02:
What I meant is that there should be a permanent box-selection watcher. 
When it detects that box selection should be started :
- If no preselection: Start box selection
- If an object is preselected : Querry the object if it is OK to start box selection.

This way if the object is a measurement label or if it is a solid of an active assembly : don't start box selection and instead start dragging it.

Then do you still see a problem @wwmayer ?

marcuspollio, 2024-12-02:
Hi there !

Please see https://github.com/FreeCAD/FreeCAD/issues/13477#issuecomment-2323536377 (though the list is not quite exhaustive).

Sorry if the following sounds obvious...
IMHO and from experience in BIM or other FOSS softwares, I don't mind having explicit Selection commands if they can be grouped into a single drop-down in a global Toolbar, _Edit_ or right-click contextual menus (e.g. single selection, range selection, "combo" box and lasso selections, plus any modifiers like addition or subtraction).

In most of them, box selection is also the default behavior called only using the mouse (or trackpad/tablet/whatever) without any command, in a click-and-drag fashion, from an "empty" location (not hovering an specific element like a vertex, label or gizmo). This usually works consistently, whatever the mode (e.g. edit or not) and editor or UI widget (2D, 3D, spreadsheet, etc.) When hovering a specific element, the latter is generally transformed (e.g. translation) by a similar click-and-drag. A pre-selected element/object is also transformed by a subsequent drag.

Having the above in FreeCAD would be the easiest from a user perspective.
I sadly cannot comment on the implementation side of it =D

wwmayer, 2024-12-02:
> Then do you still see a problem

Pre-selection is optional and can be switched off by the user. And for huge models it's already planned to automatically switch off and on the pre-selection to speed up the rendering.

In such cases the box selection may jump in when it shouldn't.

PaddleStroke, 2024-12-03:
This sounds like a very corner case to me.
And in this case we can simply deactivate box selection at the same time as preselection.

MarcelP102, 2024-12-06:
As a user I find it not really user friendly to press a button to start box selection instead of click and drag the mouse. All the paid CAD software I use at the office works this way (AutoCAD/Vectorworks/Revit(BIM)/SketchUp etc..). I hope this will implemented in Freecad. Would love to donate some money for this task.